EttusResearch/uhd #271: Timed frequency switching on multiple channels not possible

Jacky2k, 2019-04-23:
When switching the frequency using timed commands in a multi-channel system (X310, TwinRX) on a single timestamp for all channels, it fails without giving an error. When putting small delays between each set_command_time() before each set_rx_freq() it works just fine.
The effect of this is very strange. The received waveform is more or less correct, but the spectrum is swapped (so I guess I and Q are swapped) for every second channel. I attached a small example program to show this effect. You can set the define USE_PAUSE to 1 to enable a delay of 1 ms between the set_rx_freq() which will make the program work as expected.
The attached images show the first and the second measurement. Also after setting the frequency a second time, everything works as expected.

To compile the program you need to install some libs first:
sudo apt install libgnuplot-iostream-dev libarmadillo-dev
The program can be compiled with:
g++ -std=c++14 main.cpp -luhd -lboost_system -larmadillo -lboost_iostreams -lboost_filesystem -o main
I use an USRP X310 with two TwinRX, each fed with the same (using a -24dB power splitter) 100 MHz signal at -40 dBm (so -64 dBm at each receiver input). The USRP is connected over 10G ethernet.
UHD is "linux; GNU C++ version 5.5.0 20171010; Boost_105800; UHD_3.15.0.git-89-gf93c5227" from yesterday. FPGA image is also from yesterday using the download script - where can I find the version number? I am running an up-to-date Ubuntu 16.04.

[main.cpp.zip](https://github.com/EttusResearch/uhd/files/3106645/main.cpp.zip)
[image]
[image]

Jacky2k, 2019-04-26:
We just found out that even switching the frequency one after another does not fully get rid of the problem. Currently we see a random 180° phase shift of every second channel, but only for certain gain values of the USRP. This is very strange, I know, and probably cannot be reproduced by anyone without significant effort, but maybe it may help to figure out what is going on.

michael-west, 2019-07-16:
@Jacky2k The 3.14.1.0 release includes a few fixes for TwinRX, including one related to timed commands.  Please try that version and let us know if the problem is resolved.

natetemple, 2019-07-26:
@Jacky2k We have reproduced this on 3.14.1.0. We have an issue on our internal bug tracker so I'm going to close this issue. We hope to have a fix for this on an upcoming UHD release.

closed 2019-07-26 by natetemple