EttusResearch/uhd #273: IQ-correction values changed with factor of 2

Kuurusch, 2019-04-24:
## Issue Description ##
Between the UHD-version 3.14-rc1 and the final release 3.14.0.0, the IQ-correction-values (amplitude-correction and phase-correction) are now quite exactly the half than before. Has this a reason or is it a bug?

## Setup Details ##
We use UHD 3.14.0.0 and the appropriate FPGA-Image for the x310 USRP with UBX-160 Daughterboards.

## Expected Behavior ##
We would expect to have the same correction-values than in UHD 3.14-rc1

## Actual Behaviour ##
Amplitude and Phase correction values are both the half than before.

## Steps to reproduce the problem ##
We apply the correction with our own app, using the UHD-functions set_rx_iq_balance()

Kuurusch, 2019-05-03:
@Ettus: Any sugestions? Why are the values divided with factor 2 compared to your older release?

mbr0wn, 2019-05-03:
We have an idea why this is happening, but we're not quite sure what to do about it. One thing I'm not sure of: Is this a problem for you, or do you just need to know the rationale? It's possible that for 3.15, we'll go back to the values we had for RC1. I realize it's not an ideal outcome, and it was not an anticipated issue.

Kuurusch, 2019-05-06:
@mbr0wn The change is not serious for us but would be good to know, if it will stay like this or if it could change back in future releases. Because we have calibrated devices out in the field and if we update them, we have also to calibrate them again.

mbr0wn, 2021-06-29:
There's not much we can do here, so I'm closing this. However, we're more closely tracking output power going forward to avoid this happening again. We might not be able to avoid it happening, though -- we'll try to use release notes for this kind of thing.

closed 2021-06-29 by mbr0wn