EttusResearch/uhd #274: Help on touse for UHD3.9 LTS

Zethernos, 2019-04-24:
## Issue Description ##
Trouble to utilize UHD3.9 on Ubuntu 18.04.

## Setup Details ##
Hi everyone, I am trying to build srsLTE, although I am supposed to ask for help in srslte or softwaresystems forums, however, i was unable to find the solution as the posts need to be verified by the moderator and I have waited for about several days.
I have tried by uninstallling and reinstalling, it didn't work.
I am totally new to Linux and srsLTE. Besides that the UHD error code: 11 may I know what does it mean? It does show work on my both computers especially building with b210 from ettus.
However, the other computer Ubuntu 16.04 with b210, I managed to get it to UHD 3.9.7.

## Expected Behavior ##
Instead of running of uhd 3.15 I would like to use uhd 3.9.7

## Actual Behaviour ##
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.15.0.git-84-g164d76dc
Opening USRP with args: 
Error opening UHD: code 11
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 2.50 Mhz
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz
set TX frequency to 2684999936
set RX frequency to 2564999936
Setting Sampling frequency 11.52 MHz
Set RX sampling rate 11.52 Mhz, filter BW: 10.00 Mhz

## Steps to reproduce the problem ##
 git clone https://github.com/EttusResearch/uhd.git 
cd uhd 
git checkout UHD-3.9.LTS 
cd host 
mkdir build 
cd build
cmake .. 
make 
sudo make install 
sudo ldconfig 
## Additional Information ##
If there is anything I can add in, I will try to reply you asap. Thanks in advance!

Zethernos, 2019-04-24:
However, after building once with the commands, I keep encountering errors with 
Error opening UHD:Code 11
No compatible RF frontend found
Error opening RF device
Failed to find device auto with args auto

The only solution is by restarting the computers again, then I won't be able to encounter the errors. Is there another method?

mbr0wn, 2019-04-29:
@Zethernos These error messages are not coming from UHD, so we can't really tell you what's going on. Based on the error message, I would recommend checking your device is actually capable of doing whatever you're trying to do.
I'm closing this issue for now. If you think this is a bug in UHD, please reopen!

closed 2019-04-29 by mbr0wn