EttusResearch/uhd #275: x310 - uhd 14_0 rm_multi_samples based examples throws error only recovered by power

ilayni, 2019-04-24:
## Issue Description ##

Hi i am working with X310, 2 twin rx, uhd 14_0
GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.14.0.0-0-unknown.

I am running an example which sample in continuous mode 4 channels based on rx_multi_samples and writing the result to 4 files

at the end i get 
[ERROR] [UHD] Exception caught in safe-call.
  in ctrl_iface_impl<_endianness>::~ctrl_iface_impl() [with uhd::endianness_t _endianness = (uhd::endianness_t)0u]
  at /home/gig/uhd/host/lib/rfnoc/ctrl_iface.cpp:52
this->send_cmd_pkt(0, 0, true); -> EnvironmentError: IOError: Block ctrl (CE_04_Port_70) no response packet - AssertionError: bool(buff)
  in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0u; uint64_t = long unsigned int]
  at /home/gig/uhd/host/lib/rfnoc/ctrl_iface.cpp:142

after this happens only power up enables communicating with usrp
uhd_usrp_probe doesn't work as well

## Setup Details ##

Hi i am working with X310, 2 twin rx, uhd 14_0
GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.14.0.0-0-unknown.

## Expected Behavior ##

samples with no errors
## Actual Behaviour ##

erros
## Steps to reproduce the problem ##

try running and stop capturing in the middle

none
## Additional Information ##

2997215859, 2019-06-06:
I meet the same error. And I cannot transmmit data when the firmware version is  36. If the firmware version is 33, I can transmmit the data.

michael-west, 2019-06-07:
Thank you for posting.  We are currently testing UHD v3.14.1.0-rc1 and will add this to the set of tests.  I believe the simple workaround is to revert to UHD v3.13.1.0 while we investigate.

Can you post the exact commit hash you are using?

2997215859, 2019-06-10:
> Thank you for posting. We are currently testing UHD v3.14.1.0-rc1 and will add this to the set of tests. I believe the simple workaround is to revert to UHD v3.13.1.0 while we investigate.
> 
> Can you post the exact commit hash you are using?

gnuradio commit hash: 9e04b27bc4c96f95215b727ba3320812faa8d6aa
uhd commit hash: f83faf28b3424ba60c3bdc40d408011c9c619c8a

ilayni, 2019-06-10:
Hi current
UHD v3.14.1.0-rc1

Has the same problem.
V 3.13.1 has a different bug which requires power cycle of the board.

michael-west, 2019-06-10:
Thank you for the updated information.  We will test on v3.14.1.0-rc1.

michael-west, 2020-10-12:
Presumed to be resolved in UHD 4.0.0.0.  Please re-open if the issue persists or contact support@ettus.com if a fix is required on an earlier version.

closed 2020-10-12 by michael-west