EttusResearch/uhd #278: X310 twin RX

mrobertsn, 2019-05-05:
## Issue Description ##
X310 with 2x twinRX cannot stream >1MSPS without reporting overflows forcing it to stop.

## Setup Details ##
Have been using v3.13.1 relatively problem free (at least this issue). upgraded to v3.14 and anything above 1MSPS it stops and reports overflows. This is for both a discrete and continuous stream)

HAve replicated this with a b205mini and the mini works fine with both versions.

## Expected Behavior ##
stream >1MSPS.

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

mbr0wn, 2019-05-22:
@mrobertsn We were not able to reproduce this. Even 100 Msps works fine. Is it possible that your disk is too slow? Can you try with `--null`? I'll close this, unless there's more info.

mrobertsn, 2019-05-22:
Hi Martin,
The definitely has not been resolved, we have tried the above and it didn't make a difference. The issue didn't exist in the previous version 3.13.1 but occurs now. I would be interested to know what system you tested yours on, was it just linux running gnu-radio, as we created a .exe file and are running it under windows.

mbr0wn, 2019-05-22:
Yup, running on Linux.

mbr0wn, 2019-05-22:
Is this specific to TwinRX, or the X310 (on UDP)? Do you happen to have an N210, or N310?

mrobertsn, 2019-05-22:
I have 2x twin rx running in a x310

mrobertsn, 2019-05-22:
> Yup, running on Linux.

Perhaps its a windows issue. I noticed there were a lot of changes re the twin rx etc in UHD 3.14, did the twin rx running uhd 3.14 even get tested by ETTUS under windows before the release?

Same-Writer, 2019-07-26:
Follow up to @mrobertsn , I was able to perform a handful of successful tests on my Windows 10 machine at 10Msps RX on 4 channels with TwinRX. I'll post my console output below. I'm working over a single 10GbE link with my Jumbo Packet size set to 9014 (Windows NIC setting). Beyond that, everything is stock Windows 10 with UHD 3.14.1.0 in use.

```
C:\Program Files\UHD\lib\uhd\examples>benchmark_rate.exe --args addr=192.168.30.2 --rx_rate 10e6 --rx_channels "0,1,2,3"

[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_107000; UHD_3.14.1.0-release
[00:00:00] Creating the usrp device with: addr=192.168.30.2...
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
[WARNING] [UDP] The MTU (1472) is larger than the FastSendDatagramThreshold (1024)!
This will negatively affect the transmit performance.
See the transport application notes for more detail.

[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1311 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1320 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
[WARNING] [X300] Cannot update master clock rate! X300 Series does not allow changing the clock rate during runtime.
Using Device: Single USRP:
  Device: X-Series Device
  Mboard 0: X310
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: TwinRX RX0
  RX Channel: 1
    RX DSP: 1
    RX Dboard: A
    RX Subdev: TwinRX RX1
  RX Channel: 2
    RX DSP: 0
    RX Dboard: B
    RX Subdev: TwinRX RX0
  RX Channel: 3
    RX DSP: 1
    RX Dboard: B
    RX Subdev: TwinRX RX1
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: Unknown (0x0094) - 0
  TX Channel: 1
    TX DSP: 0
    TX Dboard: B
    TX Subdev: Unknown (0x0094) - 0

[00:00:05.264688] Setting device timestamp to 0...
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
[00:00:07.053212] Testing receive rate 10.000000 Msps on 4 channels
[00:00:17.461526] Benchmark complete.

Benchmark rate summary:
  Num received samples:     398720808
  Num dropped samples:      0
  Num overruns detected:    0
  Num transmitted samples:  0
  Num sequence errors (Tx): 0
  Num sequence errors (Rx): 0
  Num underruns detected:   0
  Num late commands:        0
  Num timeouts (Tx):        0
  Num timeouts (Rx):        0

Done!
```

If you're still experiencing issues after setting your system up in the same way described above, please reach out to support@ettus.com. We can open a ticket and perform troubleshooting specific to your setup. I do not believe this is a generalized bug with UHD on Windows.

closed 2019-07-26 by natetemple