EttusResearch/uhd #279: Assertion error in get_freq_and_freq_word () in dsp_core_utils.cpp

ruvimk, 2019-05-10:
## Issue Description ##
Simply calling uhd::usrp::multi_usrp::make () crashes sometimes when the application is built in Release configuration from the C++ UHD code. It always works correctly in Debug configuration. This is a memory initialization issue caused by the _current_freq property being set initially to a garbage value in tx_dsp_core_3000_impl and passed tx_dsp_core_3000_impl::set_freq () from tx_dsp_core_3000_impl::set_tick_rate (), because set_tick_rate () re-does the set_freq () using _current_freq, before _current_freq was even initialized to anything. 

To fix this issue: 
In tx_dsp_core_3000.cpp, at the end of the tx_dsp_core_3000_impl class where it defines some private variables, add an "= 0" to the definition of _current_freq. Just in case the other properties have this issue, can add "= 0" to the others as well. So, instead of this: 
`private:
    wb_iface::sptr _iface;
    const size_t _dsp_base;
    double _tick_rate, _link_rate;
    double _scaling_adjustment, _dsp_extra_scaling, _host_extra_scaling, _fxpt_scalar_correction;
    double _current_freq;`

Should do this: 
`private:
    wb_iface::sptr _iface;
    const size_t _dsp_base = 0;
    double _tick_rate = 0, _link_rate = 0;
    double _scaling_adjustment = 0, _dsp_extra_scaling = 0, 
		_host_extra_scaling = 0, _fxpt_scalar_correction = 0; 
    double _current_freq = 0;`

## Setup Details ##
Using the latest FPGA images from (https://files.ettus.com/binaries/images/?C=M;O=D), which as of this writing, is January 30, 2018. This is image major version 14. The latest host C++ code that supports this image version is commit ID 82aef3c from the GitHub (use 'git checkout 82aef3c'). Compiled with Visual Studio 2017 with some minor compiler issue fixes to the code. 

## Expected Behavior ##
The function uhd::usrp::multi_usrp::make () should make a USRP device. 

## Actual Behaviour ##
When running in Debug mode, it always works. When running in Release mode, this still works, but only most of the time. However, 1 out of every 7-15 times running the Release version, the application throws an assertion error: 

> Creating the usrp device with: vid=0x2500,pid=0x0020...
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.1; Boost_106600; UHD_3.12.0.arete-80-gb14bca13
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO....
[INFO] [GPS] Found an internal GPSDO: LC_XO, Firmware Rev 0.929a
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
Error: AssertionError: std::abs(freq) <= tick_rate/2.0
  in void __cdecl get_freq_and_freq_word(const double,const double,double &,int &)
  at C:\Users\rkondratyev\Documents\ODIN\cpp\uhd\host\lib\usrp\cores\dsp_core_utils.cpp:29

## Steps to reproduce the problem ##
Command line: 
`rx_samples_to_file --channel="0,1" --freq=90100000 --continue --duration=3 --args=vid=0x2500,pid=0x0020`

## Additional Information ##
The host/examples/rx_samples_to_file.cpp compiled code.

mbr0wn, 2019-05-21:
@ruvimk Thanks so much for the report. I wasn't able to reproduce this, but your logic is easy to follow. The default values you suggest need some improving, but we'll have a patch up for this soon.

closed 2019-05-22 by mbr0wn

drmpeg, 2019-06-03:
Just wanted to chime in and say that I'm also seeing this issue on my B210. It's amazing that it hasn't shown up before. For me, GNU Radio 3.8 works fine, but 3.7.13.5 fails.

Just setting _current_freq to 0 works for me.

mbr0wn, 2019-06-03:
@drmpeg Are you using the same version of UHD with both those versions of GNU Radio?

drmpeg, 2019-06-03:
@mbr0wn Yes, I'm using v3.14.0.0 for both. I also saw the problem on v3.13.0.3-rc1.

This kind of behavior is not unusual for uninitialized variables. Small changes in memory layout cause different effects.

In my testing, the assert always happened when _current_freq ended up being a NaN.

mbr0wn, 2019-06-04:
@drmpeg Thanks for the feedback. We have a fix on master, and we'll cherry-pick that onto UHD-3.14 before the next stable release.