EttusResearch/uhd #280: X310 time stamping appears to be broken in STREAM_MODE_NUM_SAMPS_AND_DONE

markosli, 2019-05-13:
When receiving samples in STREAM_MODE_NUM_SAMPS_AND_DONE I have observed timing deviations of as much as several hundred microseconds. The problem seems to be exacerbated by the occurrence of 'O's while receiving (more likely to occur).

In order to help debug this issue, I have created a reduced executable which exhibits the issue (source attached) using no equipment external to the X310. 

Compiling & running the example file attached shows that in continuous mode, the PPS TRIG OUT of the X310  when fed into an LFRX appears to have very consistent rising edge timing. However, when running in num_samps_and_done mode, the rising edge of the PPS appears to occasionally be off by as much as 100uS.

Instructions for how to compile the code, and run in either continuous or num_samps_and_done mode are included in the source.
[edge_detector.zip](https://github.com/EttusResearch/uhd/files/3173914/edge_detector.zip)

mbr0wn, 2019-05-22:
@markosli Thanks for providing the easy-to-replicate example.

michael-west, 2020-10-12:
@markosli There is a bug in your example code that would account for the inaccuracy.  When using NUM_SAMPS_AND_DONE, there is a gap in time between when the end of the burst is processed and the new stream command is issued.  If the PPS edge occurs during that time, your example with report the rising edge as the first sample in the new burst, which will be some random amount of time away from the actual edge.  To remedy it, please align the start of a burst so it is before the expected PPS edge and make sure the burst is long enough to capture the edge.

This does not look like a bug to me, so I am closing it for now.  If you make the changes and still have the issue, please feel free to re-open it.  Thank you!

closed 2020-10-12 by michael-west

markosly, 2020-12-11:
Hi Michael, Replying from alternate account here. I do not quite follow the explanation of the bug.

Line 120: 
num_samps_this_recv = rx_stream->recv(buffs, max_samps_per_packet, md, recv_timeout);

populates buffs and md.

Even though the new stream command is issued a few lines later (124), this should not modify the contents of buffs and md before the rising edge detection for-loop is run. Thus, given that buffs and md have not been modified yet, and given that they are aligned with each other I believe the issue remains. Am I missing something?