EttusResearch/uhd #283: time sync to gpsdo fails, /sync_to_gps  example fails, usrp time is too slow(half of what is should be)

ilayni, 2019-05-23:
## Issue Description ##

This bug is reproduced with ettus eample ~/uhd/host/build/examples$ ./sync_to_gps 
tested on x310 with gpsdo uhd ver 3.14.00 output attached.\

x 310 with 2 twin rx connected to gpsdo
i set time source and clock source to gpsdo
and than sync usrp time to it
for example if gps time is 1558515300 - and i set it to the usrp
after 10 second time read from usrp should be
1558515310
but it actually is
1558515305
usrp is tick seems to be to slow, 0.5 of the rate it should be.

## Setup Details ##

UHD 14.00 original example fails to sync_to_gps.
This is with checking the original examples shipped
with ettus official release of  uhd ver 3.14.00
running  uhd/host/build/examples$ ./sync_to_gps
Here is the output
This looks like a bug in UHD 3.14.0
------

~/uhd/host/build/examples$ ./sync_to_gps 

Creating the USRP device with: ...
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.14.0.0-0-unknown
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
[INFO] [GPS] Found an internal GPSDO: LC_XO, Firmware Rev 0.929a
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1319 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1315 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
Using Device: Single USRP:
  Device: X-Series Device
  Mboard 0: X310
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: TwinRX RX0
  RX Channel: 1
    RX DSP: 1
    RX Dboard: A
    RX Subdev: TwinRX RX1
  RX Channel: 2
    RX DSP: 0
    RX Dboard: B
    RX Subdev: TwinRX RX0
  RX Channel: 3
    RX DSP: 1
    RX Dboard: B
    RX Subdev: TwinRX RX1
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: Unknown (0x0094) - 0
  TX Channel: 1
    TX DSP: 0
    TX Dboard: B
    TX Subdev: Unknown (0x0094) - 0

Synchronizing mboard 0: X310

**************************************Helpful Notes on Clock/PPS Selection**************************************
As you can see, the default 10 MHz Reference and 1 PPS signals are now from the GPSDO.
If you would like to use the internal reference(TCXO) in other applications, you must configure that explicitly.
You can no longer select the external SMAs for 10 MHz or 1 PPS signaling.
****************************************************************************************************************

Waiting for reference lock...LOCKED
WARNING:  GPS not locked - time will not be accurate until locked
USRP time: 1558515302.999999762
GPSDO time: 1558515304.000000000

ERROR: Failed to synchronize USRP time to GPS time

## Expected Behavior ##

be able to sync to gps, usrp samples time correctly
## Actual Behaviour ##

usrp is half as slow as it should be
getting time from usrp shows it incremnets in 1 second every 2 seconds
## Steps to reproduce the problem ##

run uhd/host/build/examples$ ./sync_to_gps 

## Additional Information ##

Same-Writer, 2019-06-28:
This looks to be a regression introduced in [v3.14.0.0-rc1](https://github.com/EttusResearch/uhd/commits/v3.14.0.0-rc1). We've identified a fix that'll be included in the next tagged release, but anything prior to the rc1 tag shouldn't exhibit this behavior.

michael-west, 2020-10-12:
Resolved in UHD 4.0.0.0.

closed 2020-10-12 by michael-west