EttusResearch/uhd #284: B210 - Error receiving from UHD :30

mcclochette, 2019-05-24:
## Issue Description ##

I want to run a reliability, durability test. So, what I want to do is run my program as long as I would like it to run.
The issue is it crashes between 3 and 50 minutes. I never made it work past 50 minutes. The data that I am sending is just "ping" packets. 
The error that I received is from the UHD. It is an UHD_ERROR_IO (error 30).
When I looked online about what this issue is, it is said that it is raised whenan I/O operation fauls for an I/O related reason :S.
Is it my cable ? Is it my usb3 adaptor ? Is it my usrp ?

## Setup Details ##

I'm using two usrp b210 with srslte. One is for the UE, the other one is for the ENB. I have an attenuator in the middle (45db). My computer is running Ubuntu 18.04 LTS, UHD 3.14.0 (libuhd, uhd-host, uhd003), srslte 19.03. 

## Expected Behavior ##

I want to run a reliability, durability test. So, what I want to do is run my program as long as I would like it to run.

## Actual Behaviour ##

The error that I received is from the UHD. It is an UHD_ERROR_IO (error 30).
The program cannot run after that. It crashes. 

## Steps to reproduce the problem ##

Start 

## Additional Information ##

michael-west, 2019-06-04:
@mcclochette The error does indicate some type of issue on the USB transport and is being reported by UHD, but it is difficult to say what the cause could be.  It could be in the srsLTE software or in UHD.  To eliminate srsLTE, try running the benchmark_rate and tx_bursts UHD examples using the same parameters (i.e. sample rates) as the application to see if the problem occurs with just UHD.  The benchmark_rate example will make sure the system can handle the continuous sample rate.  The tx_bursts will test repeated bursts.  If the error reproduces with those, post the commands run and the exact errors seen so we can reproduce the issue and dig in further.

mcclochette, 2019-06-04:
@michael-west Thank you for responding! We had access to another usrp b210, so we change it with the new one. The error went away. Before that, we tried with a different USB cable, without success. We will look into your solution to debug the problem later. Thanks again.

closed 2019-06-04 by mcclochette