EttusResearch/uhd #290: Race condition shows itself as permissions error during B200 initialization

karl-marl, 2019-07-01:
## Issue Description ##
Although appropriate system rules are installed, UHD host tools (for e.g. `uhd_find_devices` or `b2xx_fx3_utils`) still **most of the times** trigger some *permissions error* when initializing a just plugged-in device.
Despite this error, the device seems eventually *somewhat* initialized, as it's correctly enumerated.
Subsequent calls to `uhd_find_devices` or `b2xx_fx3_utils` do not trigger the error.  

Increasing from 1 second to 3 seconds the sleep delay that occurs after firmware loading (in `b200_iface.cpp`) reliably removes these errors, which would suggest *some* race condition.

## Setup Details ##
OS: Linux 5.1.15 (Fedora 30)
CPU: i7-6700HQ CPU @ 2.60GHz
UHD: 3.15.0.git-1-gf83faf28
USRP: B200 (VID:2500 PID:0020)

The setup above is my current setup.
But IIRC, I've observed a very similar behavior testing a large number of Linux kernel and UHD driver versions.
And AFAICT, the code path that may include the race condition is *not even specific to Linux*. 

## Expected Behavior ##
Once appropriate permissions are set at system level (for e.g. udev rules on Linux), UHD tools should not trigger any *permissions error*.

## Actual Behaviour ##
For e.g. `uhd_find_devices`:
```
$ uhd_find_devices 
[INFO] [UHD] linux; GNU C++ version 9.1.1 20190503 (Red Hat 9.1.1-1); Boost_106900; UHD_3.15.0.git-1-gf83faf28
[INFO] [B200] Loading firmware image: /usr/local/share/uhd/images/usrp_b200_fw.hex...
[ERROR] [USB] USB open failed: insufficient permissions.
See the application notes for your device.

--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    serial: 30xxxxx
    name: myb200
    product: B200
    type: b200
```
For e.g. `b2xx_fx3_utils`:
```
/usr/local/lib64/uhd/utils/b2xx_fx3_utils -D
Device opened (VID=0x2500,PID=0x0020)
B2xx detected... Control of B2xx granted...

Loading firmware
[INFO] [UHD] linux; GNU C++ version 9.1.1 20190503 (Red Hat 9.1.1-1); Boost_106900; UHD_3.15.0.git-1-gf83faf28
[INFO] [B200] Loading firmware image: /usr/local/share/uhd/images/usrp_b200_fw.hex...
Cannot open device
```

## Steps to reproduce the problem ##
1. boot host
2. login
3. plug-in B200
4. run `uhd_find_devices`
5. note the `[ERROR] [USB] USB open failed: insufficient permissions.` message
6. note that the device is eventually enumerated

or

1. boot host
2. login
3. plug-in B200
4. run `b2xx_fx3_utils`
5. note the `Cannot open device` message

## Additional Information ##
In file `uhd/host/lib/usrp/b200/b200_iface.cpp` we read:
```
// TODO: Replace this with a polling loop in the FX3, or find out
// what the actual, correct timeout value is.
std::this_thread::sleep_for(std::chrono::milliseconds(1000));
```
If we increase the `sleep_for()` delay from 1 second to 3 seconds, UHD tools do **NOT** trigger any permission error (on *my* system).
This, and the fact the error is triggered *only most of the times*, suggest some kind of *race condition* during device initialization.

My concern is: I could probably live with this (at the end of the day, the device is enumerated, despite the error messages), **but** it is always enumerated (and always do operate) at USB2 speed (not at Super Speed). Which is not suitable for a large number of applications.

So, this report may be:
* an *issue* (read a bug descritpion), as there's somewhat a lack of synchronization that (improperly) shows itself as a permission error
* a request for some light: could this (supposed) race condition have any side-effect that may prevent the B200 to *operate* at USB3 speed ?

Thanks.

natetemple, 2019-07-26:
@karl-marl Thanks for the detailed report!

Could you please email into support@ettus.com ? We can help further debug the issue there.

I'm going to close this issue and move it to our internal bug tracker.

closed 2019-07-26 by natetemple

karl-marl, 2019-07-26:
@natetemple I'd be more than contended if you could help me regarding which USB **3** chipset / Linux kernel / UHD driver versions we should assume compatibles with B200-like hardware. Once upon a time, there was a link from the Ettus website to a list of *supported* hardware and systems (OS), that's dead, for years now.
As I said, I'd be very glad, and will send some email to support@ettus.com to request advice (and would be equally glad to reach/read the *internal bug tracker* you raised).

Though, I'm not sure why me opening some kind of ticket about my Linux/USB3 troubles,  may relate to you closing this issue about a race condition in `b200_iface.cpp`.
AFAICT, this issue i) is emphasized by some comments within the corresponding source code, and ii) may also show itself when one (as a UHD driver user) don't even think about it (for e.g., it shows itself as some *permission* error).

So, for me, it seems that closing this issue, *without* solving the race condition, or whatever you name it, shifts a technical question to a support ticket.

And worst, this:
* may prevent others to find this thread (since it's *closed*), and understand they do **NOT** face any *permission* related error (for e.g. to prevent peoples to double-check their udev rules, or AppArmor / SEL policies, till the end of the night)  
* hides the technical issue into the source code itself, without any other artefact to help developers and users memory

Which is not nice, IMHO. As you well know.

Thanks.

michael-west, 2020-10-12:
@karl-marl This sounds like it is an issue with the re-enumeration that the Cypress FX3 does after loading the firmware.  We have released a bootloader for the B200 which may resolve the issue.  To load the bootloader:

- Download the bootloader image using `uhd_images_downloader -t b2xx`
- Load the bootloader using `b2xx_fx3_utils -B <install_dir>/share/uhd/images/usrp_b200_bl.img`

Give that a try and re-open this issue if it does not fix it.

liumaolin-931, 2021-05-20:
Hi
I have encountered a similar problem, but I am using b210, inserting usrp into uhd_find_devices will display "USB open failed: insufficient permissions. see the application notes for your device." but I can find the device normally using "sudo uhd_find_devices". I am using GNURadio to play usrp, and this problem occurs when I run GRC streaming,But I don’t want to use "sudo" permissions.