EttusResearch/uhd #293: so many UHD status: L= U= S=. and TRX discontinuity too wide

kovalevskySPB, 2019-07-25:
## Issue Description ##
We get a lot of messages UHD Status in terminal on the system Fedora 27. On the site Ettus says that there is not enough processor performance, but on servera rather efficient processor is inserted and the load does not exceed 10%; this is verified in the system monitor and in the top -H output. After a certain number of messages the error: TRX discontinuity too wide ... ms and everything crashed. This situation happens when we use USRP x310. Tried to use different MTU values, as well as Jumbo frames.
To solve this problem, UHD versions from 3.9.0 to 3.14.1 have been tried.

## Setup Details ##
linux; GNU C++ version 7.3.1 20180712 (Red Hat 7.3.1-6); Boost_106400; UHD_3.12.0.HEAD-0-gec786351
uname -a:
Linux [name] 4.18.19-100.fc27.x86_64 #1 SMP Wed Nov 14 22:04:34 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
USRP x310

michael-west, 2019-07-25:
@kovalevskySPB Thank you for posting.  UHD has no "TRX discontinuity too wide" error message, so that must be coming from some other software component.  'L' indicates a late packet on transmit, 'U' indicates and underrun (Radio needed data and none had arrived), and 'S' indicates a sequence error which usually occurs if packets are being dropped on the network.  I recommend checking the throughput on your network interface.

This is a support issue and not a bug, so it is being closed.  Please contact support@ettus.com for assistance with your specific use case or post on the [usrp-users mailing list](http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com).

closed 2019-07-25 by michael-west

kovalevskySPB, 2019-07-29:
> @kovalevskySPB Thank you for posting. UHD has no "TRX discontinuity too wide" error message, so that must be coming from some other software component. 'L' indicates a late packet on transmit, 'U' indicates and underrun (Radio needed data and none had arrived), and 'S' indicates a sequence error which usually occurs if packets are being dropped on the network. I recommend checking the throughput on your network interface.
> 
> This is a support issue and not a bug, so it is being closed. Please contact [support@ettus.com](mailto:support@ettus.com) for assistance with your specific use case or post on the [usrp-users mailing list](http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com).

Thank you for your fast reply! We use the Amarisoft software, but their technical support could not help us. They said that this error relates to UHD. This error appears in the ENB section, after this software stops its work. The connection between the USRP and the server is organized by 1 Gbit / s Etnernet. A data stream of more than 10 mega samples per second is guaranteed to give the error described above, given that 16 bits are contained in the sample. We inspected the used TFP modules and cables Etnernet. Everything is good.

arzulo, 2021-08-10:
> Thank you for your fast reply! We use the Amarisoft software, but their technical support could not help us. They said that this error relates to UHD. This error appears in the ENB section, after [...]

Sorry to revive a very old thread, but I'm currently using Amarisoft software and experiencing this same issue.  Do you remember what you did to fix the problem?

viranetw, 2021-10-05:
> Sorry to revive a very old thread, but I'm currently using Amarisoft software and experiencing this same issue. Do you remember what you did to fix the problem?
I am using the Amarisoft software and getting the same issue. If you had overcome this issue can you please the steps to fix the problem?

viranetw, 2021-10-05:
> Sorry to revive a very old thread, but I'm currently using Amarisoft software and experiencing this same issue. Do you remember what you did to fix the problem?

Hello I am also using the Amarisoft software and have encountered the same problem. If you solved the problem can you please provide the steps to fix the problem?