EttusResearch/uhd #294: UHD 3.15.0.git-0-g6563c537 E310 radio channel 0 produces only noise

ChromeBumper, 2019-07-26:
## Issue Description ##
Radio Port 0 produces nothing but noise

## Setup Details ##
UHD 3.15.0.git-0-g6563c537
multiple E310 SG1 devices

## Expected Behavior ##
simultaneous signals from both radio channels

## Actual Behaviour ##
Radio Channel 1 produces expected signal data, Channel 0 produces nothing but noise regardless of which radio port that channel is routed to

## Steps to reproduce the problem ##
generate and install SD image e3xx_e310_sg1_sdimg_default-v3.15.0.0-e310_prerelease.zip from http://files.ettus.com/binaries/cache/e3xx/meta-ettus-v3.15.0.0-e310_prerelease/ into an E310 SG1
connect antennas two all four ports.
Run 
usr/lib/uhd/examples/rx_samples_to_file --freq=915e6 --rate=1e6 --gain=76 --channel=0 --file=0.out
and then 
usr/lib/uhd/examples/rx_samples_to_file --freq=915e6 --rate=1e6 --gain=76 --channel=1 --file=1.out
0.dat contains noise only, 1.dat contains signal data
re-run with different gain levels: 0.dat contains noise of level similar to that in 1.dat, but no signal -- as if there were no antenna connected.
swap antennas and repeat (no change)
repeat with fresh build using accompanying sdk and git pull, install in ~/newinstall (no change)
repeat with rfnoc_rx_to_file (no change)
modify rfnoc_rx_to_file to swap ports A and B with radio ports 0 and 1

    stream_args.args["block_id0"] = radio_ctrl->get_block_id();
    stream_args.args["block_port0"] = "1";
    stream_args.args["block_id1"] = radio_ctrl->get_block_id();
    stream_args.args["block_port1"] = "0";

Note that it is still radio channel 0 that's dead, and that good data is now coming in using RF port "A"

Repeat in a custom FPGA build without DDC -- same result

Change gain -- noise level changes on both channels, but Channel 0 produces only noise (as if the antenna were disconnected)

swap SD card with one made using http://files.ettus.com/e3xx_images/e3xx-release-4/ 
or earlier -- both channels work with those images (but those UHD versions have RFNoC issues)

mbr0wn, 2019-11-25:
Using a mini-siggen.

## Streaming into B-side, 76 dB gain:

[image]

## Streaming to B-side, 10 dB gain

[image]

The level is way down.

## Streaming to A-side, 10 dB gain

mbr0wn, 2019-12-04:
Thanks for the report -- a fix will come very soon to the UHD-3.15.LTS and master branches.

closed 2019-12-04 by mbr0wn