EttusResearch/uhd #296: Timeouts on N310 - Stemming from liberio/dma

mcarosino, 2019-08-11:
## Issue Description ##
When running a host -> Rfnoc -> host type flowgraph, after a somewhat random period of time "timeout on chan 0" starts occurring and never recovers (no further data is ever received on the host).

## Setup Details ##
N310 with UHD v3.13.0.2 and its associated FPGA tag.
Latest Liberio commit
GNURadio installed on the N310 and flowgraph is run in embedded mode (e.g. ssh into the n310 and run the python script directly - the problem probably occurs in remote mode as well due to the nature of it but I've not tested it much).

## Expected Behavior ##
Flowgraph should run indefinitely (although intermittent timeouts might happen, it should not just stop receiving data permanently)

## Steps to reproduce the problem ##
Run any flowgraph in embedded mode like the following - I chose a ddc/duc combo as those are in the default image, but a fifo->fifo will give the same result.
[image]

## Additional Information ##
After some debugging, it appears the host stops receiving data from the rfnoc side because the transmit side of the flowgraph hangs. I traced the issue through the following call chain: work_tx_u -> super_send_pack_handler: send() -> send_one_packet() -> convert_to_in_buff -> buff.reset() -> ( see attached stack trace) ....  -> liberio_zero_copy_mrb: get_new() -> liberio_chan_buf_deque() -> select()

It's pretty obvious with printf debugging that we are continually sending packets in the above path and then randomly one of the calls hangs and never returns. This appears to happen in the call to select() in liberio_chan_buf_deque() never timing out (I haven't figured out why this is the case, as it's called with a 0.1 sec / 100000 microsec timeout value. I also tried manually modifying the timeout value and that does not fix the issue so I suspect its more likely an issue with the usrp dma kernel driver than the code calling select)

Attached is a gdb backtrace taken of the flowgraph after it starts timeout (and I manually stop it with ctrl-c). Thread 10 is the thread that is hung at select().

[n310_gdb_stacktrace.txt](https://github.com/EttusResearch/uhd/files/3490329/n310_gdb_stacktrace.txt)

mcarosino, 2019-08-11:
Okay minor update, further tracing and it appears the issue is manifesting itself in tx_flow_ctrl function in device3_io_impl.cpp 

The tx_flow_ctrl function has a while(true) loop that continually tries to grab a recv buffer, extract a flow control packet, and update the free flow control window space before exiting. At some point when the problem manifests itself, the call to async_xport->get_recv_buff() always returns a null buff so we just sit in this loop continually try to get a receive buffer and never get one. (this is why I initially thought the code was hanging at select() when actually it was just repeatedly calling it and timing out).

I'm not sure if this behavior indicates an issue with the dma or just that the fpga is not sending this flow ctrl packet - should be easy enough to verify on the fpga side though..

mcarosino, 2019-08-12:
Noticed this may also be related to https://github.com/EttusResearch/uhd/issues/203

mcarosino, 2019-08-12:
I've confirmed the last sequence number sent from the Duc's flow control responder is the same as that received by the host side's tx_flow_ctrl() function - meaning that for some reason the fpga is not sending the next flow control packet as needed. The question is if this is happening due to 1) the duc block getting into a bad state, 2) the DDC block getting into a bad state and backpressuring the DUC, 3) the receive streamer running into an issue and backpressuring the whole thing.

mcarosino, 2019-08-13:
Getting closer .. looks like the host side actually thinks it sent packet with sequence number N+1, but when the issue occurs, the last sequence number seen by the FPGA flow control responder is N. So when the host goes to send another packet (seq N+2), it can't because the FPGA hasn't ack'd the previous seq # N+1 and will sit in tx_flow_ctrl waiting for it indefinitely.

I'm trying to figure out now if the missing PS->PL packet actually makes it across the DMAs or not. Also a quick work around for the issue may just be to timeout in the tx_flow_ctrl and re-send the last unack'd packet - I might give this a try.

mcarosino, 2019-08-18:
Got a resolution on this issue - problem was with the configuration of an IP core in the DMA block in the BD.

I noticed that in a flowgraph like file source -> fifo -> fifo -> file sink, occasionally the output data would randomly have what looked like a CHDR header in it. Using this as an event to trigger ILAs off of, I was able to backtrace through the axi_wrapper, noc_shell, and up to the RFNoC crossbar to clearly see a situation where the crossbar was being given data on its Host port that looked like one CHDR packet sandwiched right in the middle of another.

Looking at the tx DMA block in the BD, it's made up of 10 axi_dmac blocks that are connected to a 10:1 axis_interconnect (containing a 10:1 axis_switch). The interconnect is setup for tlast arbitration so that normally only a full , contiguous CHDR packet could be sent on the output bus to the RFNoC cross bar. However, the "arbitrate on max transfer" settings was also set to 1 - this has the effect that now there will be a new arbitration if either tlast is asserted OR tvalid is deasserted. 

So essentially you could clearly see a long CHDR packet from the host being sent to the crossbar and sometimes the sending dma would briefly deassert tvalid (for whatever reason) while at the same time another dma had already asserted tvalid indicating it was ready to transfer - at this point the interconnect granted this new arbitration, completed the smaller transfer (in this case a flow control packet from the host) and then went back to the previous packet once its tvalid was reasserted. In the worst case the timing of this would be such that the packet insertion broke up the first packets chdr header - which is probably what lead to the ultimate timeouts.

TLDR: axis_interconnect IP should be set to arbitrate on max transfers = 0.  I've opened a PR for the fpga fix.

mbr0wn, 2019-10-17:
Thanks again so much for hunting this down and providing a fix. It's been merged.

closed 2019-10-17 by mbr0wn