EttusResearch/uhd #300: man page compression should be optional

ZeroChaos-, 2019-10-01:
most linux distribution packaging software handles selectable man page compression which is broken by force compressing the man pages with gzip.  This default behavior is fine (although I'd prefer xz), but there needs to be a way to disable it so the package manager can handle it.

michaelld, 2019-10-01:
Can you test this patch? It works in my testing for both compressed (default) and uncompressed.

[patch-uhd_man_pages_optional_compress.diff.txt](https://github.com/EttusResearch/uhd/files/3677336/patch-uhd_man_pages_optional_compress.diff.txt)

ZeroChaos-, 2019-10-01:
glorious, tested with and without the flag and this appears to work and keep the original default.  Please accept this patch.

michaelld, 2019-10-01:
@ZeroChaos- Thx for your feedback. Hoping it's not too difficult to get merged.

closed 2019-10-09 by mbr0wn

ZeroChaos-, 2019-10-10:
@mbr0wn I checked a few branches, felt like at random, and didn't see this committed.  If you can tell me which branch I should be testing I'm happy to test this and https://github.com/EttusResearch/uhd/issues/301 to make sure it's all good.

michaelld, 2019-10-10:
@ZeroChaos- it's currently in a private dev repo. It'll be pushed to the public UHD repo soon-ish. It works very nicely... I think you'll be pleased!

ZeroChaos-, 2019-10-10:
I'm sure it will be fine, I just like to close the loop and test the dev code :-) you know how to find me pre-release if you want extra verification