EttusResearch/uhd #301: PKG_DOC_DIR should be overridable

ZeroChaos-, 2019-10-01:
It doesn't appear that PKG_DOC_DIR is overridable:

https://github.com/EttusResearch/uhd/blob/master/host/CMakeLists.txt#L125

In gentoo (and possibly others) we like docs in versioned directories.  I am currently moving the documents around before they get merged to the filesystem, but it would be nice to support overriding this directory.  thanks for the consideration

michaelld, 2019-10-01:
Easiest way to handle this is probably the following, changed from the noted CMakeLists.txt code ... would this work for you?
```
if(NOT DEFINED PKG_DOC_DIR)
    set(PKG_DOC_DIR share/doc/uhd)
endif()
if(NOT DEFINED PKG_MAN_DIR)
    set(PKG_MAN_DIR share/man/man1)
endif()
```

ZeroChaos-, 2019-10-01:
Yes, that would be wonderful.  I'm not sure if anyone requires the man location to be overridden, but why not? :-)

michaelld, 2019-10-01:
True. I'll remove that change from the my forthcoming PR ...

michaelld, 2019-11-13:
This issue was fixed. Closing.

closed 2019-11-13 by michaelld

ZeroChaos-, 2019-11-14:
This and my other packaging fix going into 3.15.0.0?

michaelld, 2019-11-14:
Yes! They are already in the public UHD GIT master: https://github.com/EttusResearch/uhd.git

ZeroChaos-, 2019-11-14:
thank you kind sir