EttusResearch/uhd #302: macOS dyld issue

citizenrich, 2019-10-02:
## Issue Description ##
`uhd_find_devices` fails with a dyld error.

## Setup Details ##
$ brew info boost
boost: stable 1.71.0 (bottled), HEAD
Collection of portable C++ source libraries
https://www.boost.org/
/usr/local/Cellar/boost/1.71.0 (14,377 files, 522.8MB) *
  Poured from bottle on 2019-09-08 at 19:00:22
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/boost.rb
==> Dependencies
Required: icu4c ✔
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 58,023 (30 days), 124,653 (90 days), 650,168 (365 days)
install_on_request: 19,689 (30 days), 44,170 (90 days), 208,299 (365 days)
build_error: 0 (30 days)
~
$ uhd_find_devices
dyld: lazy symbol binding failed: Symbol not found: __ZN5boost16re_detail_10700027cpp_regex_traits_char_layerIcE4initEv
  Referenced from: /usr/local/Cellar/uhd/3.14.1.0/lib/libuhd.3.14.1.dylib
  Expected in: /usr/local/opt/boost/lib/libboost_regex-mt.dylib

dyld: Symbol not found: __ZN5boost16re_detail_10700027cpp_regex_traits_char_layerIcE4initEv
  Referenced from: /usr/local/Cellar/uhd/3.14.1.0/lib/libuhd.3.14.1.dylib
  Expected in: /usr/local/opt/boost/lib/libboost_regex-mt.dylib

Abort trap: 6

michaelld, 2019-10-02:
Did you update Boost while UHD was installed, without recompiling UHD?

citizenrich, 2019-10-02:
Thanks. The brew compiled version must be out of sync with boost, so I took the homebrew boost update and then it broke uhd. I've built it myself successfully to solve the issue.

closed 2019-10-02 by citizenrich