EttusResearch/uhd #304: UHD can not detect rfx daughterboard correctly

josnchen, 2019-10-25:
when run "uhd_usrp_probe" under uhd-3.13.1, for N210 + RFX2400, it prints the error info:

[ERROR] [DBMGR] The daughterboard manager encountered a recoverable error in init.
Loading the "unknown" daughterboard implementations to continue.
The daughterboard cannot operate until this error is resolved.
LookupError: KeyError: key "0" not found in dict(i, N14adf4360_regs_t17prescaler_value_tE)

i checked the file host/lib/usrp/dboard/db_rfx.cpp. in double rfx_xcvr::set_lo_freq() about line 371.
there maybe have  a mistake when using auto keyword.  auto BS/auto P in for loop defines a local variable, but the BS/P will be used out of for loop, then the BS/P always equals to 0.  at last the error occurs.

the following is a simple solution to fix this,  maybe useful.

@@ -369,16 +369,18 @@ double rfx_xcvr::set_lo_freq(
     for(R = 2; R <= 32; R+=2){
-        for(auto BS:  bandsel_to_enum.keys()){
+        for(auto aBS:  bandsel_to_enum.keys()){
+            BS = aBS;
             if (ref_freq/R/BS > 1e6) continue; //constraint on band select clock
-            for(auto P:  prescaler_to_enum.keys()){
+            for(auto aP:  prescaler_to_enum.keys()){
                 //calculate B and A from N
+                P = aP;
                 double N = target_freq*R/ref_freq;
-                B = int(std::floor(N/P));
-                A = boost::math::iround(N - P*B);
+                B = int(std::floor(N/aP));
+                A = boost::math::iround(N - aP*B);
                 if (B < A or B > 8191 or B < 3 or A > 31) continue; //constraints on A, B
                 //calculate the actual frequency
-                actual_freq = double(P*B + A)*ref_freq/R;
+                actual_freq = double(aP*B + A)*ref_freq/R;
                 if (actual_freq/P > 300e6) continue; //constraint on prescaler output
                 //constraints met: exit loop
                 goto done_loop;

mbr0wn, 2019-11-22:
@josnchen Thanks so much for the bug report, and tracking down the issue. I have submitted a PR that is a bit more in vain of the original code and it is going through internal CI and review.

The final fix will look something like the attached one.

[0001-rfx-Fix-calculation-of-prescaler-and-band-select.patch.txt](https://github.com/EttusResearch/uhd/files/3881213/0001-rfx-Fix-calculation-of-prescaler-and-band-select.patch.txt)

closed 2019-11-22 by mbr0wn

jcs3, 2020-03-19:
I am experiencing this problem with uhd-3.13.1 and an USRP N200. I have actually tried multiple N200s and multiple daughtercards - no combination works correctly.

Has this problem been fixed?  If so, in what version of UHD?
Thanks in advance