EttusResearch/uhd #311: x3x0 TwinRx setting sample rate after creating a rx_streamer throws "RuntimeError: Multiple sampling rates upstream of RX Terminator 0: ..."

craigtong, 2019-11-21:
Since 3.15 I am no longer able to set the rx_rate after a rx_streamer has been created.

When channels on both RX slots of the x3x0 are used it would appear that only the first DDC rate is set but not the second and a check then fails. For example, setting 25 MSps (where 20 MSps was set successfully before the creation of rx_streamer) throws:

RuntimeError: Multiple sampling rates upstream of RX Terminator 0: RuntimeError: Node 0/DDC_0 specifies 2.5e+07, node 0/DDC_1 specifies 2e+07.

I first noticed this in my custom application but I have reproduced it in rx_multi_samples but adding another usrp->set_rx_rate() after the streamer is created (I've attached this modified file, see line 130).

[rx_multi_samples.zip](https://github.com/EttusResearch/uhd/files/3873429/rx_multi_samples.zip)

Any combination of channels where the DDCs on both sides are of the board are initiated causes the exception:

./rx_multi_samples --channels="0,1,2,3"
./rx_multi_samples --channels="0,1,2"
./rx_multi_samples --channels="0,2"
./rx_multi_samples --channels="0,2,3"
./rx_multi_samples --channels="1,2"
etc.

Combinations of channels where initialisation of the DDC is on 1 side only work.
./rx_multi_samples --channels="0,1"
./rx_multi_samples --channels="2,3"

My setup is GNU C++ version 7.3.0; Boost_106500; UHD_3.15.0.0-0-gdbaf4132 on Gentoo Linux x64 kernel 4.14.65
Tried with both X310 and X300 with 2x TwinRXs.

This behavior was not a problem with any previous UHD versions so I'm assuming this is new bug, unless the streamer needs to be destroyed and recreated between rate changes but that seems unlikely?

steviez, 2019-12-02:
Hi @craigtong,

Thanks for bringing this to our attention and and apologies on the delayed response; much of the team is based out of the US and we had a holiday (Thanksgiving) last week which made last week a slow week on our end.

> unless the streamer needs to be destroyed and recreated between rate changes but that seems unlikely?

As far as I understand, I agree that we shouldn't need to destroy the streamer and this looks like an unintentional change of behavior / bug.

I began looking at this and was able to both:
- Recreate the problem behavior with an even newer UHD 3.15 commit than what you used AND
- Confirm that the issue did not exist in [UHD 3.14.1.1](https://github.com/EttusResearch/uhd/releases/tag/v3.14.1.1)

I have identified the problem / commit that caused this change of behavior; however, I still need to discuss and review potential fixes with my peers. Look for an update within the next couple days

steviez, 2019-12-02:
Hi @craigtong 

> I have identified the problem / commit that caused this change of behavior; however, I still need to discuss and review potential fixes with my peers. Look for an update within the next couple days

After discussing the issue with a coworker, we actually feel that the error is the proper result. When we have a steamer object, all channels for that streamer must have the same rate. The process of updating the rate once the streamer has been created occurs channel by channel. As such, midway through the update, the streamer is in a bad state. The reason why you didn't see this in 3.14.1.1 or before is that we actually didn't have proper error checking in place. This error checking is new to our 3.15 branch, which is why you wouldn't have encountered this before.

>As far as I understand, I agree that we shouldn't need to destroy the streamer and this looks like an unintentional change of behavior / bug.

So back-pedaling on my previous statement, at this point, we do recommend destroying / creating a new streamer object if you need to change the rate. Given that you suggested this yourself, I'm guessing you'd know how to go about updating your application.

Does this all make sense / do you have a path forward with moving your application to 3.15?

craigtong, 2019-12-03:
Hi @steviez 

Thank you for looking into the issue.

While I can certainly make this work I find the dependency a little bit strange at least philosophically speaking.

To me it seems like being able to change the sample rate (something attributable to the span of a spectrum analyser) seems like it should be a runtime setting just like gain or centre frequency in an SDR and therefore not require a major re-initialisation of objects.

We have a UHD wrapper class hierarchy that implements various radar specific functionality. The tx/rx streamer is created on construction of the class given things like channel selection, packing specification etc. Centre frequency, sample rate, and channel gain are then mutators of the class at runtime.

I would ask why not check for the consistent sample rates after all channels have been set (at least for a logical DDC block) rather that one at a time. I also wonder how this maps to behaviour in complex sampling setups like with the UBX?

Nontheless I have not familiarised myself with the inner workings of the rx_streamer class and I trust there are good reasons for the current architecture.

Given that a sample rate change comes with the caveat of having to re-prime the timestamp engine at the new rate (which has a turnaround time of next PPS + another PPS anyway) it is really not a deal breaker within our software design to also recreate the streamer object. The difference will be negligible.

closed 2019-12-03 by craigtong

ilayni, 2020-01-09:
Hi
Same problem here
Can you post an example code of how to solve it ?
I get 
Exception set_rx_rateRuntimeError: Multiple sampling rates upstream of RX Terminator 0: RuntimeError: Node 0/DDC_0 specifies 2.5e+07, node 0/DDC_1 specifies 49019.6.

craigtong, 2020-01-09:
Hi @ilayni 

You need to create the rx_streamer object after you call set_rx_rate. And recreate the rx_streamer object after each successive call to set_rx_rate.

ilayni, 2020-01-09:
Craig, thank you
TKs
Trying to do it, without success, 
Example code?

steviez, 2020-01-09:
@ilayni - See the `rx_multi_samples.cpp` example:
https://github.com/EttusResearch/uhd/blob/master/host/examples/rx_multi_samples.cpp

If you examine main(), you'll see the following:
```
    ...

    // set the rx sample rate (sets across all channels)
    std::cout << boost::format("Setting RX Rate: %f Msps...") % (rate / 1e6) << std::endl;
    usrp->set_rx_rate(rate);

    ...

    // create a receive streamer
    // linearly map channels (index0 = channel0, index1 = channel1, ...)
    uhd::stream_args_t stream_args("fc32"); // complex floats
    stream_args.channels             = channel_nums;
    uhd::rx_streamer::sptr rx_stream = usrp->get_rx_stream(stream_args);

    ...
```

Your application will need to follow this order of operations like @craigtong mentioned, which may include destroying / re-creating a streamer when you want to change your sample rate.

ilayni, 2020-01-12:
HI, Steve, Craig tks for the useful info
I destroy the streamer by calling reset 
My issue is since i use 4 channels, 
4 threads are created and i get a race condition where rxStream_ equals zero
if i try to use a mutex, i get locked.

					for(i=0;i<numChannels_;i++)
							channel_nums.push_back(i);
					curRate=0;
					uhd::stream_args_t rx_stream_args("sc16","sc16");
					rx_stream_args.channels = channel_nums;
					rxStream_.reset();
					rxStream_ = usrp_->get_rx_stream(rx_stream_args);

craigtong, 2020-01-13:
Hi Ilayni

The user code to receive 4 channels should require only 1 thread. A single call to recv() will pack your buffers with all 4 channels. Furthermore, as I recall, rx_streamer's functions are not thread safe so it sounds like your issue may be related to that.

Perhaps post more code so that we can get a better idea of what we are trying to do.

ilayni, 2020-01-15:
Race condition was solved, issue now is that if i destroy and create rx_streamer
after doing it 62 times i get:

Exception EnvironmentError: IOError: Block ctrl (CE_01_Port_40) no response packet - AssertionError: bool(buff)
in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0u; uint64_t = long unsigned int]
at /home/gig/uhd/host/lib/rfnoc/ctrl_iface.cpp:151

jjgudugu, 2021-07-02:
Hi ilayni,
I want to use 4 channels to receive simultaneously, but with different central frequency and sampling rate. I met the same problem "RuntimeError: Multiple sampling rates upstream of RX Terminator 0: RuntimeError: Node 0/DDC_0 specifies 1e+07, node 0/DDC_1 specifies 5e+06." when trying to set different sampling rates. After reading all this page and still don't know how to do it. Any advise?

ilayni, 2021-07-02:
Hi
If I remember correctly, this happened exactly after switching rates 356
times
There was only one version. I think that worked for me
I think it was 14.1

jjgudugu, 2021-07-04:
Hi ilayni. Thanks a lot, I'll try this version.