EttusResearch/uhd #313: make fails on macOS Catalina 10.15 if source stored in /tmp/ directory

bayandin, 2019-11-26:
## Issue Description ##

make fails on macOS Catalina (10.15) if source stored in **/tmp** directory: 
```
cd /tmp/uhd-20191126-3334-1f7tlnl/uhd-3.14.1.1/host/build/lib && /usr/local/bin/python3 /tmp/uhd-20191126-3334-1f7tlnl/uhd-3.14.1.1/host/lib/transport/nirio/lvbitx/process-lvbitx.py --output-src-path=/tmp/uhd-20191126-3334-1f7tlnl/uhd-3.14.1.1/host/build/lib/transport/nirio/lvbitx --uhd-images-path=/usr/local/Cellar/uhd/3.14.1.1/share/uhd/images /tmp/uhd-20191126-3334-1f7tlnl/uhd-3.14.1.1/host/lib/transport/nirio/lvbitx/x310.lvbitx_base
Traceback (most recent call last):
  File "/tmp/uhd-20191126-3334-1f7tlnl/uhd-3.14.1.1/host/lib/transport/nirio/lvbitx/process-lvbitx.py", line 194, in <module>
    with open(os.path.join(autogen_src_path, class_name + '_lvbitx.hpp'), 'w') as source_file:
FileNotFoundError: [Errno 2] No such file or directory: '../../../../../../../tmp/uhd-20191126-3334-1f7tlnl/uhd-3.14.1.1/host/build/lib/transport/nirio/lvbitx/x300_lvbitx.hpp'
make[2]: *** [lib/transport/nirio/lvbitx/x300_lvbitx.cpp] Error 1
```
Full log: [uhd.log](https://github.com/EttusResearch/uhd/files/3891695/uhd.log)

It looks like the reason for the issue is Python behaviour on Catalina: https://bugs.python.org/issue38295 which affects generation of *_lvbitx.hpp (https://github.com/EttusResearch/uhd/blob/v3.14.1.1/host/lib/transport/nirio/lvbitx/process-lvbitx.py#L33).

Replacing `os.path.relpath` with `os.path.realpath` helps to fix it on Catalina, but I don't know how this change could affect other systems and configurations. If you're ok with I'll be happy to create a PR with it.

Ref: https://github.com/Homebrew/homebrew-core/pull/47101

## Setup Details ##
macOS Catalina (10.15)
UHD  3.14.1.1 / 3.15.0.0-rc2

michaelld, 2019-11-26:
why?!!! I commented on the brew issue ... bring it here too: There are other os.path.relpath in that script `host/lib/transport/nirio/lvbitx/process-lvbitx.py` . Do those also need to be changed? Seems to me that using the absolute path would work cleanly, so why use the relative path? hmmm ...

michaelld, 2019-11-26:
OK so replacing all `relpath` with `realpath` works for me not in `/tmp`. Can you test / verify that it works for you too?

michaelld, 2019-11-26:
I'll note that I'm still using Mojave ... but I do have a Catalina build computer setup that I can bring to life if necessary for testing ... just thinking if someone is on the other end here this would be faster ...

bayandin, 2019-11-26:
> so why use the relative path?

It seems because of https://github.com/EttusResearch/uhd/commit/9517de45709adaea8b574011573a565007149d5d.

So we need to test it on Windows (hope it was "fixed" since 26 Oct 2016).

> OK so replacing all `relpath` with `realpath` works for me not in `/tmp`. Can you test / verify that it works for you too?

It works for on Catalina as well

michaelld, 2019-11-26:
@bayandin Thanks for the info & the answer to "why?" ... I was mostly being dramatic, but I do really appreciate your answer!

I'll discuss with Ettus R&D and we'll see if we can come up with a way to make this work for both macOS Catalina and Windows. Fun times ahead, I'm sure!

bayandin, 2019-11-26:
> l discuss with Ettus R&D and we'll see if we can come up with a way to make this work for both macOS Catalina and Windows. Fun times ahead, I'm sure!

Thanks!

As an ugly solution which doesn't require much testing on Windows, I can suggest something like `if platform.system() == 'darwin'`

michaelld, 2019-11-26:
Yeah that could work. I'm hoping that `realpath` will work on modern Windows ... it's been a while since that commit was made, so hopefully newer Windows and Python will work! One step at a time ...

michaelld, 2019-11-26:
This change looks good in Windows 32 and 64 using MSVC 2015; still waiting for MSVC 2017. Also builds on all Linux variants thus far. So ... looking positive at the moment!

michaelld, 2019-12-01:
Looks like `realpath` works on all of our Windows CI. I'll fixup my PR to be legit & see what sort of discussion ensues.

bayandin, 2019-12-02:
That's great, thanks @michaelld!

michaelld, 2020-01-16:
FYI this change has been merged into our internal UHD repo, and will be public at the next release. I'm going to close this issue as fixed. Thanks for reporting!

closed 2020-01-16 by michaelld

bayandin, 2020-01-16:
Thanks @michaelld!