EttusResearch/uhd #314: X310 FPGA Compatibility issue with 3.15 LTS version

gnss-tao, 2019-11-27:
## Issue Description ##
UHD Version:
The latest 3.15 LTS version (hash: 035292d8ac83f8abe4e0092f03f8327d4e1f178b)

FPGA Version:
http://files.ettus.com/binaries/cache/x3xx/fpga-42c9b9b/x3xx_x310_fpga_default-g42c9b9b.zip

Device:
X310

when I run the command:  .\uhd_usrp_probe.exe  --args="addr=192.168.10.2"
I got the following error:
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.1; Boost_107100; UHD_3.15.0.HEAD-0-0887012a
[INFO] [X300] X300 initialization sequence...
Error: RuntimeError: Expected FPGA compatibility number 36, but got 35:
The FPGA image on your device is not compatible with this host code build.
Download the appropriate FPGA images for this version of UHD.

I am wondering which fpga binary should I use to flash x310.

BR

natetemple, 2019-11-28:
Hi @gnss-tao,

You should run `uhd_images_downloader` then `uhd_image_loader` to update the FPGA image.

Also, this is not a bug, so I'm going to close this issue. Please email support@ettus.com for any support issues.

closed 2019-11-28 by natetemple

gnss-tao, 2019-11-28:
[image]

@natetemple
Hi, I followed the install sequence. The screen shot shows what I did. But stiil no wrong fpga version.

craigtong, 2019-12-05:
@gnss-tao 

The image downloader script is downloading new firmware to D:\vcpkg\packages\uhd_x64-windows... and the flashing script is fetching firmware from D:\Program Files\UHD...

You are likely flashing old firmware to the device. You probably need to set UHD_IMAGES_DIR accordingly to ensure coherence between paths.

gnss-tao, 2019-12-09:
@craigtong 
I see. Thanks a lot. That's the problem. 
Now it is working well.

hassanmaree, 2023-09-06:
Hi all, whenuhd find devices i get fpga : n/a? moreover, i cannot acces gnu radio from the root of ni usrp e312. can any 1 please help me?