EttusResearch/uhd #315: [B210] Signal spikes when TX power is changed at runtime

cfr34k, 2019-11-28:
## Issue Description ##

When the TX power setting of the B210 is changed while a signal is transmitted,
spikes with very high level occur in the output signal. It seems like the gain is briefly set to maximum.

I noticed this first while experimenting with the `gain` message for the GNU
Radio USRP sink, but could also reproduce it with `uhd_siggen_gui`.

Some screenshots/plots are attached to illustrate the issue:

- [gr_usrp_sink_toggle_50dB_20dB](https://user-images.githubusercontent.com/1255682/69822591-09ff0380-1207-11ea-959c-9c5dacfaff53.png) shows a measurement of the signal generated by GNU Radio. Trace 1 (yellow) shows the signal at 50 dB gain. I wrote a script that repeatedly toggles the gain between 50 dB and 20 dB via the USRP Sink's command message port and accumulated the signal peaks via the spectrum analyzer's peak hold function (trace 2, cyan). What emerges after a few minutes looks like an amplified version of the transmitted signal, but with very bad phase noise.
- [gr_time_domain_20_to_50dB](https://user-images.githubusercontent.com/1255682/69822590-09ff0380-1207-11ea-9024-add5cb26fb47.png) shows a time-domain recording of this signal while changing from 20 to 50 dB gain. The first part is at 20 dB gain, the overdriven part is the unexpected high-power signal and the final part is the normal level at 50 dB gain.
- [uhd_siggen_sinus_20_to_50dB](https://user-images.githubusercontent.com/1255682/69822592-0a979a00-1207-11ea-95eb-aaa0c30efea0.png) shows a similar setup, but the signal is generated using `uhd_siggen_gui` this time. This time I manually moved the gain slider through various values, but never above 50 dB. Even though there are fewer spikes in the spectrum due to manual testing, the same shape emerges.

The peak power in the trace 2 spectrum from GNU Radio is very close to the power that the USRP generates at maximum gain setting. Therefore, I suspect that the USRP briefly switches to maximum gain while configuring a new setting.

## Setup Details ##

- Hardware: USRP B210
- OS: Ubuntu 18.04.3 LTS
- UHD Version: 3.14.1.1 (as released on Github)

## Expected Behavior ##

When requesting a change in TX gain, the output signal power is adjusted accordingly without spikes or glitches. Also, the power should never what one expects from the gain values before and after the change.

## Actual Behaviour ##

The output power massively exceeds the expected power for a short time (~100 samples @ 3MSps) before the new gain is applied.

## Steps to reproduce the problem ##

- run `uhd_siggen_gui -g 20 -f 1530000000`
- monitor/record the output signal
- adjust the _TX Gain_ from the GUI
- spikes as described above will appear in the analog signal

Gain changes that always triggered the spikes on my setup are 20<->50 dB and 23<->26 dB in both directions.

## Additional Information ##

If you need something else, please let me know.

mbr0wn, 2021-09-16:
This is really easy to reproduce. Here, using uhd_siggen_gui and uhd_fft in scope mode, triggering on an amplitude I should never be seeing:
[image]

mbr0wn, 2021-09-16:
Fix is a reorder of the register writes. I'm putting this into our internal CI pipeline:
```patch
From 6e6114d9657fb4b052d3f2f523846bcb62bedcc8 Mon Sep 17 00:00:00 2001
From: Martin Braun <martin.braun@ettus.com>
Date: Thu, 16 Sep 2021 13:33:23 +0200
Subject: [PATCH] ad9361: Modify set-tx-gain procedure to update gain in one go

The previous behaviour of UHD for setting gain was:
1. Set "Mask Clr Atten Update". This will avoid "Immediately Update TPC
   Atten" to be cleared.
2. Then, assert "Immediately Update TPC Atten".
3. Poke the LSBs of the attenuation value.
4. Poke the MSB of the attenuation value.

This order of operations has the downside of causing large Tx power
spikes when setting the attenuation, because you need both registers to
properly set the attenuation, but we are updating the gain immediately,
even between the two attenuation register's update.

Moreover, the upstream Linux driver for AD9361 by ADI also does not
do this. We therefore change the procedure to match the kernel driver
behaviour, which is:

0. [During initialization: Clear "Mask Clr Atten Update"
1. Poke the attenuation registers
2. Then, assert "Immediately Update TPC Atten".

This avoids Tx power spikes. It also reduces the Tx-gain procedure to
3 pokes instead of 4.
---
 .../common/ad9361_driver/ad9361_device.cpp    | 20 ++++++++++++-------
 1 file changed, 13 insertions(+), 7 deletions(-)

diff --git a/host/lib/usrp/common/ad9361_driver/ad9361_device.cpp b/host/lib/usrp/common/ad9361_driver/ad9361_device.cpp
index 9275c83f5..5574022b6 100644
--- a/host/lib/usrp/common/ad9361_driver/ad9361_device.cpp
+++ b/host/lib/usrp/common/ad9361_driver/ad9361_device.cpp
@@ -1817,6 +1817,10 @@ void ad9361_device_t::initialize()
 
     /* Set TXers & RXers on (only works in FDD mode) */
     _io_iface->poke8(0x014, 0x21);
+
+    /* We won't be using immediate-Tx-attenuation, so we de-assert "Mask Clr
+     * Atten Update". */
+    _io_iface->poke8(0x077, 0x00);
 }
 
 void ad9361_device_t::set_io_iface(ad9361_io::sptr io_iface)
@@ -2203,11 +2207,6 @@ double ad9361_device_t::set_gain(direction_t direction, chain_t chain, const dou
 
         return gain_index;
     } else {
-        /* Setting the below bits causes a change in the TX attenuation word
-         * to immediately take effect. */
-        _io_iface->poke8(0x077, 0x40);
-        _io_iface->poke8(0x07c, 0x40);
-
         /* Each gain step is -0.25dB. Calculate the attenuation necessary
          * for the requested gain, convert it into gain steps, then write
          * the attenuation word. Max gain (so zero attenuation) is 89.75.
@@ -2215,8 +2214,8 @@ double ad9361_device_t::set_gain(direction_t direction, chain_t chain, const dou
          * "value" is out of bounds, so range checking must be performed
          * outside this function.
          */
-        double atten      = AD9361_MAX_GAIN - value;
-        uint32_t attenreg = uint32_t(atten * 4);
+        const double atten      = AD9361_MAX_GAIN - value;
+        const uint32_t attenreg = uint32_t(atten * 4);
         if (chain == CHAIN_1) {
             _tx1_gain = value;
             _io_iface->poke8(0x073, attenreg & 0xFF);
@@ -2226,6 +2225,13 @@ double ad9361_device_t::set_gain(direction_t direction, chain_t chain, const dou
             _io_iface->poke8(0x075, attenreg & 0xFF);
             _io_iface->poke8(0x076, (attenreg >> 8) & 0x01);
         }
+
+        /* After we've set both registers for the new gain, we apply it (assert
+         * "Immediately Update TPC Atten"). This avoids the attenuation from
+         * intermediately becoming very small (i.e., high output power) during
+         * the set-gain cycle.
+         */
+        _io_iface->poke8(0x07c, 0x40);
         return AD9361_MAX_GAIN - ((double)(attenreg) / 4);
     }
 }
-- 
2.25.1
```

closed 2021-09-17 by atrnati