EttusResearch/uhd #318: mismatched versions of uhd and fpga images

joau-mms, 2019-12-19:
## Issue Description ##

uhd_images_downloader.py downloads FPGA images, which are not compatible with the installed uhd-host driver. Which results in following error message if one wants to use the SDR.

RuntimeError: RuntimeError: Expected FPGA compatibility number 14, but got 16:
The FPGA build is not compatible with the host code build.
Please run:

 "/usr/lib/x86_64-linux-gnu/uhd/utils/uhd_images_downloader.py"

The problem is, uhd_images_downloader.py downloads the FPGA image with version 16, while the driver the script ships with is on version 14.

The Problem is present with UHD from Ettus' Ubuntu Repo (https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd) and also if it is compiled from git Repository, Branch 3.14

## Setup Details ##

Ubuntu 18.04
UHD: 3.14.1 from
Repo: https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd
SDR: Ettus B200

## Expected Behavior ##

That uhd_images_downloader.py only gets FPGA images, which are compatible to the driver the downloader was shipped with.

## Actual Behaviour ##

compatibility version mismatch: driver has v14 and fpgs image v16

## Steps to reproduce the problem ##

1. Install driver from mentioned launchpad Repo or compile it from source (branch UHD-3.14, other branches are not tested)
2. run uhd_images_downloader as root
3.run uhd_fft -f 2400 -> run into runtime error "RuntimeError: Expected FPGA compatibility number 14, but got 16"

joau-mms, 2019-12-19:
## Update for Branch UHD 3.15.LTS
Just compiled branch UHD-3.15.LTS and I got the same error message (see below), what happens:

1. uhd_images_downloader -> runs fine
2. uhd_image_loader -> runs fine
3. uhd_fft -f 2400 -> throws error

# uhd_fft -f 2400
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
-- Detected Device: B200
-- Operating over USB 3.
Traceback (most recent call last):
  File "/usr/bin/uhd_fft", line 500, in <module>
    main()
  File "/usr/bin/uhd_fft", line 483, in main
    tb = uhd_fft(args)
  File "/usr/bin/uhd_fft", line 128, in __init__
    self.setup_usrp(uhd.usrp_source, args)
  File "/usr/lib/python2.7/dist-packages/gnuradio/uhd/uhd_app.py", line 173, in setup_usrp
    channels=self.channels,
  File "/usr/lib/python2.7/dist-packages/gnuradio/uhd/__init__.py", line 122, in constructor_interceptor
    return old_constructor(*args)
  File "/usr/lib/python2.7/dist-packages/gnuradio/uhd/uhd_swig.py", line 2683, in make
    return _uhd_swig.usrp_source_make(*args)
RuntimeError: RuntimeError: Expected FPGA compatibility number 14, but got 16:
The FPGA build is not compatible with the host code build.
Please run:

 "/usr/lib/x86_64-linux-gnu/uhd/utils/uhd_images_downloader.py"

joau-mms, 2019-12-19:
Tested 3.14.1 and 3.15.LTS with an N210 and it all runs without a problem

natetemple, 2019-12-19:
@joau-mms This issue is likely due to you having multiple UHD installations on your host. Please email support@ettus.com and we can help you further though that channel.

closed 2019-12-19 by natetemple

joau-mms, 2019-12-20:
You are right, gnuradio provided by Ubuntu 18.04 standard repositories installs and uses libuhd003.010.003. The standard repositories are also providing the packages uhd-host and uhd-soapysdr with compatible versions.
The conflict occurs if one  uses the Ettus Repo (https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd) which installs a newer version of libuhd, the uhd_* tools and newer firmware and fpga images. The same holds true for self compiled uhd software.

Right now, I'm a nit-picker, but the Ubuntu repository of Ettus research are hosting a packages, which has dependency issues (see below). The packages provided by you are suggesting installation of gnuradio from Ubuntu repos which will also install the conflicting, old libuhd version.

:~$ sudo apt show uhd-host
Package: uhd-host
Version: 3.14.1.1-0ubuntu1~bionic1
Priority: optional
Section: science
Source: uhd
Maintainer: Ettus Research <packages@ettus.com>
Installed-Size: 24,7 MB
Depends: adduser, libuhd3.14.1 (= 3.14.1.1-0ubuntu1~bionic1), python, python-requests, libboost-filesystem1.65.1, libboost-program-options1.65.1, libboost-serialization1.65.1, libboost-system1.65.1, libboost-test1.65.1, libboost-thread1.65.1, libc6 (>= 2.15), libgcc1 (>= 1:4.0), libncurses5 (>= 6), libstdc++6 (>= 5.2), libtinfo5 (>= 6)
Recommends: curl, procps, python-tk
**Suggests: gnuradio**

Anyhow, thank you very much. Your hint helped me a lot and I which you a wonderful holly days and a happy new year

passoswell, 2021-10-01:
Hello.
@joau-mms , did you solve your problem, find a solution or any lead?

I have the same issue as yours, apparently two different versions of  uhd-host driver intalled, which are UHD_003.010.003.000-0-unknown and 3.15-release.
When I run srsRAN eNB, it uses UHD_003.010.003.000-0-unknown and returns the error indicated in this issue.
When I run OpenAirInterface eNB, it uses 3.15-release and I find no issue communicating with the b240.
The problem is that even after purging uhd-host with "sudo apt-get purge uhd-host" the two frameworks can find the respective uhd drivers.

mcan06, 2021-12-11:
Have you solved this problem?

passoswell, 2021-12-13:
Hello @mcan06,

I was using the version of srsRAN that install using apt (ubuntu 20). It defaults to installing and using UHD_003.010.003.000-0-unknown. I also had installed UHD drivers from source, and then I had conflicts.

I solved my issue by removing srsRAN and all dependencies installed using apt. I also completely removed UHD drivers (apt version and manually compiled version).

Next I compiled from source and installed a version I wanted to test os UHD drivers ([this tutorial](https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux#Update_and_Install_dependencies), except for the GNU radio section), and after that I also compiled from source and installed srsRAN. This way it detects and uses the correct driver and fpga files.