EttusResearch/uhd #320: Can't build N310 FPGA image

MichaelFay, 2020-01-11:
## Issue Description ##
When building N310_XG FPGA image on ubuntu 18.04 LTS with Vivado 2018.3, build fails due to Bank VCC conflits.

## Setup Details ##
- Install Ubuntu 18.04 LTS
- Install Vivado 2018.3
- Source Vivado settigns64.sh
- Source setupenv.sh
- run "make N310_XG"

## Expected Behavior ##
N310_XG bin files should be generated. 

## Actual Behaviour ##
See attached screenshot.

MichaelFay, 2020-01-11:
[image]

a-will, 2020-01-13:
This looks like the behavior resulting from a bug in Vivado 2018.3--Only Vivado 2018.3.1 is supported. If you haven't, you will need to download and install Update 1.

MichaelFay, 2020-01-14:
All, 

After updating to 2018.3.1 I am still seeing this same issue. I am also seeing this on master and v3.15.0.0 LTS. 

Is there any known images for the N310 that are either 3.15 or master? 

-Michael

Same-Writer, 2020-01-14:
What version of UHD were you initially using when you reported this issue?

MichaelFay, 2020-01-14:
When compiling the fpga-src? 

We currently have v3.15.0.0 LTS checked out as well as master. Is it required to have uhd installed on the computer that we are using to compiling the fpga-src using Vivado? 

Also, we are able build E310 from the checkout code. 

-Michael