EttusResearch/uhd #321: unrecoverable Code crash x310, Fpga exceptio while rx capture - uhd 3.15.0 -  IOError: Block ctrl (CE_01_Port_40) no response packet -

ilayni, 2020-01-13:
## Issue Description ##
running 
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.15.0.HEAD-0-gaea0e2de
and doing rx capture with 4 channels with timed command i get the following
Exception EnvironmentError: IOError: Block ctrl (CE_01_Port_40) no response packet - AssertionError: bool(buff)
  in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0u; uint64_t = long unsigned int]
  at /home/gig/uhd/host/lib/rfnoc/ctrl_iface.cpp:151

## Setup Details ##
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.15.0.HEAD-0-gaea0e2de
and doing rx capture with 4 channels 2 twinrx 
## Expected Behavior ##
capture with no isssue

## Actual Behaviour ##
FPGA exception

## Steps to reproduce the problem ##
recv with 4 channels with time commands 

## Additional Information ##

Same-Writer, 2020-01-14:
Do you have any code that reproduces this behavior? Does it happen in v3.14.1.1?

ilayni, 2020-01-15:
I think this does not happen in 3.14.1.1

ilayni, 2020-01-15:
This happens if i create and destroy an rx_streamer
what is need in the case of changing rates in version 3.15.0

ccera-astro, 2020-04-20:
http://www.ccera.ca/files/rx_multi_samples.cpp   easily reproduces this issue -- this is a variant of the standard version that takes an --iters parameter.   I've found that the number of iterations required isn't consistent.   But what IS necessary is that the number of channels requested needs to be varying between streaming sessions.  If the channel count remains fixed, it will run to completion every time.

ilayni, 2020-04-21:
Hi,
My channel number is always 4 and i still get this issue.

atrnati, 2020-09-24:
Are you able to try to reproduce this issue with UHD 4.0.0?

michaelld, 2020-09-24:
I'm working with the customer via Ettus Support email. I'll test things out on UHD 4.0 when I get a chance soon. We're slowly working on getting UHD4 build for the customer. This issue certainly seems relevant to UHD 3.15. Let's leave this open until we get UHD4.0 tested.

Anplus, 2020-09-29:
My UHD version is 3.15.0. I didn't have such a problem when I use two x310 to implement 4 Rx or 4 Tx. But when I simultaneously use Tx and Rx, the UHD report the following errors:

thread[thread-per-block[1]: <block gr uhd usrp sink (2)>]: EnvironmentError: IOError: Failed to sync DACs! EnvironmentError: IOError: 1/Radio_0 user_reg_read64() failed: EnvironmentError: IOError: [1/Radio_0] sr_read64() failed: EnvironmentError: IOError: Block ctrl (CE_01_Port_40) packet parse error - EnvironmentError: IOError: Expected SID: 03:40>00:06  Received SID: 02:40>00:06 
thread[thread-per-block[2]: <block gr uhd usrp source (1)>]: EnvironmentError: IOError: Block ctrl (CE_01_Port_40) no response packet - AssertionError: bool(buff)
  in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0; uint64_t = long unsigned int]

cclamb, 2020-12-06:
Guys has this issue been resolved?

michaelld, 2020-12-07:
@cclamb not resolved to the best of my knowledge ... maybe the OP can re-test & report back anything related to UHD 3.15 .. or UHD 4.0 even better?

cclamb, 2020-12-09:
The error does not seem to appear on UHD 4.0.

mbr0wn, 2024-03-19:
As this no longer appears on 4.x, and we are not fixing 3.15 anymore, I'm closing this.