EttusResearch/uhd #322: Use of uninitialized memory in octoclock

argilo, 2020-01-14:
## Issue Description ##

While using valgrind to debug an unrelated issue, I noticed that octoclock uses uninitialized memory. It looks like this code fails to initialize all of `pkt_out`:

https://github.com/EttusResearch/uhd/blob/aea0e2de34803d5ea8f25d7cf2fb08f4ab9d43f0/host/lib/usrp_clock/octoclock/octoclock_impl.cpp#L100-L108

## Setup Details ##

I reproduced this by running `valgrind gqrx`. I did not have a USRP connected to my machine at the time. I was using the `v3.15.0.0` tagged release of uhd (commit aea0e2de34803d5ea8f25d7cf2fb08f4ab9d43f0) built with PyBOMBS.

## Expected Behavior ##

No valgrind errors.

## Actual Behaviour ##

```
[INFO] [UHD] linux; GNU C++ version 7.4.0; Boost_106501; UHD_3.15.0.HEAD-0-gaea0e2de
==6134== Thread 14:
==6134== Syscall param sendmsg(msg.msg_iov[0]) points to uninitialised byte(s)
==6134==    at 0x58B5607: sendmsg (sendmsg.c:28)
==6134==    by 0xB4A5B81: sendto (socket_ops.ipp:1299)
==6134==    by 0xB4A5B81: sync_sendto (socket_ops.ipp:1321)
==6134==    by 0xB4A5B81: send_to<boost::asio::const_buffers_1> (reactive_socket_service.hpp:211)
==6134==    by 0xB4A5B81: send_to<boost::asio::const_buffers_1> (datagram_socket_service.hpp:344)
==6134==    by 0xB4A5B81: send_to<boost::asio::const_buffers_1> (basic_datagram_socket.hpp:423)
==6134==    by 0xB4A5B81: udp_simple_impl::send(boost::asio::const_buffer const&) (udp_simple.cpp:51)
==6134==    by 0xB3F20CA: octoclock_find(uhd::device_addr_t const&) (octoclock_impl.cpp:107)
==6134==    by 0xB3F2755: octoclock_find(uhd::device_addr_t const&) (octoclock_impl.cpp:85)
==6134==    by 0xB1CCB0F: boost::detail::function::function_invoker1<std::vector<uhd::device_addr_t, std::allocator<uhd::device_addr_t> > (*)(uhd::device_addr_t const&), std::vector<uhd::device_addr_t, std::allocator<uhd::device_addr_t> >, uhd::device_addr_t const&>::invoke(boost::detail::function::function_buffer&, uhd::device_addr_t const&) (function_template.hpp:101)
==6134==    by 0xB4B23F9: operator() (function_template.hpp:760)
==6134==    by 0xB4B23F9: operator() (device.cpp:104)
==6134==    by 0xB4B23F9: __invoke_impl<std::vector<uhd::device_addr_t>, uhd::device::find(const uhd::device_addr_t&, uhd::device::device_filter_t)::<lambda()> > (invoke.h:60)
==6134==    by 0xB4B23F9: __invoke<uhd::device::find(const uhd::device_addr_t&, uhd::device::device_filter_t)::<lambda()> > (invoke.h:96)
==6134==    by 0xB4B23F9: _M_invoke<0> (thread:234)
==6134==    by 0xB4B23F9: operator() (thread:243)
==6134==    by 0xB4B23F9: operator() (future:1339)
==6134==    by 0xB4B23F9: std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<std::vector<uhd::device_addr_t, std::allocator<uhd::device_addr_t> > >, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<uhd::device::find(uhd::device_addr_t const&, uhd::device::device_filter_t)::{lambda()#1}> >, std::vector<uhd::device_addr_t, std::allocator<uhd::device_addr_t> > > >::_M_invoke(std::_Any_data const&) (std_function.h:302)
==6134==    by 0xB1C0EC8: operator() (std_function.h:706)
==6134==    by 0xB1C0EC8: std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) (future:561)
==6134==    by 0x58B2826: __pthread_once_slow (pthread_once.c:116)
==6134==    by 0xB4B1EFD: __gthread_once (gthr-default.h:699)
==6134==    by 0xB4B1EFD: call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()>*, bool*> (mutex:684)
==6134==    by 0xB4B1EFD: _M_set_result (future:401)
==6134==    by 0xB4B1EFD: operator() (future:1667)
==6134==    by 0xB4B1EFD: __invoke_impl<void, std::__future_base::_Async_state_impl<_BoundFn, _Res>::_Async_state_impl(_BoundFn&&) [with _BoundFn = std::thread::_Invoker<std::tuple<uhd::device::find(const uhd::device_addr_t&, uhd::device::device_filter_t)::<lambda()> > >; _Res = std::vector<uhd::device_addr_t>]::<lambda()> > (invoke.h:60)
==6134==    by 0xB4B1EFD: __invoke<std::__future_base::_Async_state_impl<_BoundFn, _Res>::_Async_state_impl(_BoundFn&&) [with _BoundFn = std::thread::_Invoker<std::tuple<uhd::device::find(const uhd::device_addr_t&, uhd::device::device_filter_t)::<lambda()> > >; _Res = std::vector<uhd::device_addr_t>]::<lambda()> > (invoke.h:95)
==6134==    by 0xB4B1EFD: _M_invoke<0> (thread:234)
==6134==    by 0xB4B1EFD: operator() (thread:243)
==6134==    by 0xB4B1EFD: std::thread::_State_impl<std::thread::_Invoker<std::tuple<std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<uhd::device::find(uhd::device_addr_t const&, uhd::device::device_filter_t)::{lambda()#1}> >, std::vector<uhd::device_addr_t, std::allocator<uhd::device_addr_t> > >::_Async_state_impl(std::tuple<uhd::device::find(uhd::device_addr_t const&, uhd::device::device_filter_t)::{lambda()#1}>&&)::{lambda()#1}> > >::_M_run() (thread:186)
==6134==    by 0x8E2966E: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25)
==6134==    by 0x58AA6DA: start_thread (pthread_create.c:463)
==6134==    by 0x97CC88E: clone (clone.S:95)
==6134==  Address 0x3a0f4385 is on thread 14's stack
==6134==  in frame #2, created by octoclock_find(uhd::device_addr_t const&) (octoclock_impl.cpp:43)
```

## Steps to reproduce the problem ##

Run `valgrind gqrx`.

argilo, 2020-01-14:
It looks like the same error occurs in many places, e.g.:

https://github.com/EttusResearch/uhd/blob/aea0e2de34803d5ea8f25d7cf2fb08f4ab9d43f0/host/lib/usrp_clock/octoclock/octoclock_image_loader.cpp#L186-L192

https://github.com/EttusResearch/uhd/blob/aea0e2de34803d5ea8f25d7cf2fb08f4ab9d43f0/host/lib/usrp_clock/octoclock/octoclock_eeprom.cpp#L36-L41

I presume this is causing uninitialized memory to be sent over the network.

mbr0wn, 2020-01-23:
Yup good catch.

closed 2020-02-04 by atrnati