EttusResearch/uhd #325: Python API: cannot stream on two channels

planewave, 2020-01-25:
## Issue Description ##

I am using a B210, with python API. When I run the example file rx_to_file.py, no problem on 1 channel, but if I want to receive date from 2 channels

python usrp_python_examples/rx_to_file.py -o foo -f 2.43e9 -d 1 -c 0 1 -n

I got the following errors

```
Traceback (most recent call last):
  File "usrp_python_examples/rx_to_file.py", line 47, in <module>
    main()
  File "usrp_python_examples/rx_to_file.py", line 39, in main
    samps = usrp.recv_num_samps(num_samps, args.freq, args.rate, args.channels, args.gain)
  File "/usr/local/lib/python3.5/dist-packages/uhd/usrp.py", line 50, in recv_num_samps
    streamer.issue_stream_cmd(stream_cmd)
RuntimeError: RuntimeError: Invalid recv stream command - stream now on multiple channels in a single streamer will fail to time align.
```
## Setup Details ##

UHD 3.15, python 3.52 and 3.76, Ubuntu 16.04
## Expected Behavior ##

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

I have emailed the support, but no reply. Then I found the solution from `usrp_python_examples/benchmark_rate.py`
This is a bug in uhd package `uhd/usrp.py` line 49
```
stream_cmd.stream_now = True
```
as the error suggested, `stream_cmd.stream_now` cannot be set to True if two channels.
it shall be replaced by 
```
stream_cmd.stream_now = (num_channels == 1)
stream_cmd.time_spec = uhd.types.TimeSpec(usrp.get_time_now().get_real_secs() + INIT_DELAY)
```
also, the `recv_buffer` need to be modified accordingly.

michaelnaw, 2020-09-14:
Same here with X310.
After applying the suggest fix, can only use rec_num_samps once. Trying to use it again causes kernel freeze (using jupyter lab)

michael-west, 2020-10-12:
@planewave @michaelnaw Thank you for reporting and providing detail.  We have opened an internal issue for further investigation.

As you pointed out, it does look like there is an issue at:  https://github.com/EttusResearch/uhd/blob/master/host/python/uhd/usrp/multi_usrp.py#L49

The only thing I can recommend in the meantime is to import uhd and use that directly.

mbr0wn, 2021-09-23:
We've applied the multi-chan fix (and some others). Note that calling `recv_num_samps()` multiple times might still fail. For this use case, we strongly recommend creating your own streamer. `recv_num_samps()` was updated to allow passing in such a streamer object.

Thanks for your report!

pfeatherstone, 2022-03-15:
I'm using the C++ API and have the same problem.
I use this:
```
stream_cmd.stream_now = channels.size() == 1;
stream_cmd.time_spec  = usrp->get_time_now() + 0.05;
```
But it times out immediately, i.e. I get the error: `uhd::rx_metadata_t::ERROR_CODE_TIMEOUT`.
Is this expected? Will it timeout until the `stream_cmd.time_spec` is reached? Or should it hang until data is available?
Thanks

mbr0wn, 2022-03-22:
@pfeatherstone does benchmark_rate work on two channels?

pfeatherstone, 2022-03-22:
@mbr0wn I'll be able to test this tomorrow. Thanks. Otherwise, do you see this kind of behaviour with dual channel streaming, i.e. loads of timeout errors? Is `recv()` still blocking when `stream_cmd.stream_now == false` and `stream_cmd.time_spec` is set to some later time? Why can't you do `stream_cmd.stream_now == true` with multiple channels?

mbr0wn, 2022-03-22:
Why can't you do `stream_cmd.stream_now == true` with multiple channels?

Because if you have a multi-channel streamer, UHD will attempt to guarantee that the time-alignment is working (i.e., the first sample on channel 0 was sampled at the same time as the first sample on channel 1). With stream_now == true, that's not possible.

pfeatherstone, 2022-03-22:
@mbr0wn Because the library doesn't permit it. I can't remember the exact error message since i don't have the SDRs here in front of me, but you definitely get an error. After some googling, you find answers that suggest using `stream_cmd.stream_now == false` for multichannel to work

pfeatherstone, 2022-03-22:
@mbr0wn Ok makes sense. Still confused about the time out messages though. It's like it's trying to flush something out.