EttusResearch/uhd #327: Build/Compile Tuple Error

TheNageek, 2020-02-04:
## Issue Description ##
Received the below error when attempting to install UHD on clean Ubuntu VM.  

```bash
In file included from /home/manager/uhd/host/lib/usrp/multi_usrp_rfnoc.cpp:16:0:
/home/manager/uhd/host/include/uhd/utils/graph_utils.hpp:30:75: error: converting to ‘std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, boost::optional<long unsigned int> >’ from initializer list would use explicit constructor ‘constexpr std::tuple<_T1, _T2>::tuple(_U1&&, _U2&&) [with _U1 = const std::__cxx11::basic_string<char>&; _U2 = const boost::none_t&; <template-parameter-2-3> = void; _T1 = std::__cxx11::basic_string<char>; _T2 = boost::optional<long unsigned int>]’
     {NODE_ID_SEP, boost::none}, {"Radio", boost::none}, {"NullSrcSink", 0}};
                                                                           ^
/home/manager/uhd/host/include/uhd/utils/graph_utils.hpp:30:75: error: converting to ‘std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, boost::optional<long unsigned int> >’ from initializer list would use explicit constructor ‘constexpr std::tuple<_T1, _T2>::tuple(_U1&&, _U2&&) [with _U1 = const char (&)[6]; _U2 = const boost::none_t&; <template-parameter-2-3> = void; _T1 = std::__cxx11::basic_string<char>; _T2 = boost::optional<long unsigned int>]’
/home/manager/uhd/host/include/uhd/utils/graph_utils.hpp:30:75: error: converting to ‘std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, boost::optional<long unsigned int> >’ from initializer list would use explicit constructor ‘constexpr std::tuple<_T1, _T2>::tuple(_U1&&, _U2&&) [with _U1 = const char (&)[12]; _U2 = int; <template-parameter-2-3> = void; _T1 = std::__cxx11::basic_string<char>; _T2 = boost::optional<long unsigned int>]’
lib/CMakeFiles/uhd.dir/build.make:1177: recipe for target 'lib/CMakeFiles/uhd.dir/usrp/multi_usrp_rfnoc.cpp.o' failed
make[2]: *** [lib/CMakeFiles/uhd.dir/usrp/multi_usrp_rfnoc.cpp.o] Error 1
CMakeFiles/Makefile2:795: recipe for target 'lib/CMakeFiles/uhd.dir/all' failed
make[1]: *** [lib/CMakeFiles/uhd.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

```

## Setup Details ##
UHD 3.15.0.0
Ubuntu 16.04.6 LTS (xenial)
gcc 5.4.0
libboost 1.58.0.1ubuntu1
cmake 3.16.3
python 2.7.12
python-mako 1.0.3
libusb-1.0-4
doxygen 1.8.11
git version 2.7.4

## Additional Information ##
It appears that the below function is the source of the issue from `graph_utils.hpp`

```c++
//! Tuple that stores a block ID, as well as an optional port number
using block_port_def = std::tuple<std::string, boost::optional<size_t>>;

// TODO: Get rid of magic strings
/*! List of blocks that can terminate chains. Note that some blocks only terminate at
 *  some of their ports, so we can optionally include a port number.
 */
static const std::vector<block_port_def> TERMINATOR_BLOCKS{
    {NODE_ID_SEP, boost::none}, {"Radio", boost::none}, {"NullSrcSink", 0}};
```

Was able to compile and I believe this would be an appropriate fix using `unordered_map` instead of tuples.

```c++
static const std::unordered_map<std::string, boost::optional<size_t>> TERMINATOR_BLOCKS
{{NODE_ID_SEP, boost::none}, {"Radio", boost::none}, {"NullSrcSink", 0}};
```

mbr0wn, 2020-02-05:
@TheNageek you mention you're on UHD 3.15, but the file you refer to only exists on master branch. Can you confirm which version you're trying?

TheNageek, 2020-02-06:
Good point, realizing that I didn't checkout a particular version.  Will give that a shot, think that will solve the problem.

TheNageek, 2020-02-06:
@mbr0wn yep that was my issue, appreciate it!

mbr0wn, 2020-02-06:
Happy to help!

closed 2020-02-06 by mbr0wn