EttusResearch/uhd #331: boost: mutex lock failed in pthread_mutex_lock while running txrx_loopback_to_file example

cjosephson, 2020-03-05:
I'm running uhd 3.15-LTS that I compiled myself. The txrx_loopback_to_file example crashes with a failed in pthread_mutex_lock when I have a transmit on one USRP and receive on another. The example runs successfully if the TX and RX device are the same. I've attached valgrind output, in case that is helpful.

Command used and output:
```
./txrx_loopback_to_file --tx-args="addr0=192.168.10.2" --rx-args="addr0=192.168.10.3" --tx-freq 1200e6 --rx-freq 1200e6 --tx-ant "TX/RX" --rx-ant "RX2" --tx-rate 2e6 --rx-rate 2e6 --wave-type SINE --wave-freq 1000 --type=float --nsamps=3000000`

Creating the transmit usrp device with: addr0=192.168.10.2...
[INFO] [UHD] linux; GNU C++ version 6.4.0; Boost_107200; UHD_3.15.0.0-7-g8d228dbe
[INFO] [USRP2] Opening a USRP2/N-Series device...
[INFO] [USRP2] Current recv frame size: 1472 bytes
[INFO] [USRP2] Current send frame size: 1472 bytes

Creating the receive usrp device with: addr0=192.168.10.3...
[INFO] [USRP2] Opening a USRP2/N-Series device...
[INFO] [USRP2] Current recv frame size: 1472 bytes
[INFO] [USRP2] Current send frame size: 1472 bytes
Using TX Device: Single USRP:
  Device: USRP2 / N-Series Device
  Mboard 0: N200r4
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: SBX/CBX RX
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: SBX/CBX TX

Using RX Device: Single USRP:
  Device: USRP2 / N-Series Device
  Mboard 0: N200r4
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: SBXv3 RX
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: SBXv3 TX

Setting TX Rate: 2.000000 Msps...
Actual TX Rate: 2.000000 Msps...

Setting RX Rate: 2.000000 Msps...
Actual RX Rate: 2.000000 Msps...

Setting TX Freq: 1200.000000 MHz...
Actual TX Freq: 1200.000000 MHz...

Setting RX Freq: 1200.000000 MHz...
Actual RX Freq: 1200.000000 MHz...

Checking TX: LO: locked ...
Checking RX: LO: locked ...
Setting device timestamp to 0...
terminate called after throwing an instance of 'boost::wrapexcept<boost::lock_error>'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted
```

However, if I run it such that the TX and RX are the same USRP (`--tx-args="addr0=192.168.10.2" --rx-args="addr0=192.168.10.3"`), then it runs without issues.

## Setup Details ##
I'm running linux kernel 4.14.52-gentoo (x86_64 Intel(R) Core(TM) i7-5650U CPU @ 2.20GHz GenuineIntel GNU/Linux), and I've built the code by checking out the 3.15-LTS branch of this repo.

I have two USRP N200 devices, probe info as follows:

```
/usr/local/bin/uhd_usrp_probe --args="addr0=192.168.10.2,addr1=192.168.10.3"
[INFO] [UHD] linux; GNU C++ version 6.4.0; Boost_107200; UHD_3.15.0.0-7-g8d228dbe
[INFO] [USRP2] Opening a USRP2/N-Series device...
[INFO] [USRP2] Current recv frame size: 1472 bytes
[INFO] [USRP2] Current send frame size: 1472 bytes
  _____________________________________________________
 /
|       Device: USRP2 / N-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: N200r4
|   |   hardware: 2576
|   |   mac-addr: 00:80:2f:0a:de:1b
|   |   ip-addr: 192.168.10.2
|   |   subnet: 255.255.255.255
|   |   gateway: 255.255.255.255
|   |   gpsdo: internal
|   |   serial: F42275
|   |   FW Version: 12.4
|   |   FPGA Version: 11.1
|   |   
|   |   Time sources:  none, external, _external_, mimo
|   |   Clock sources: internal, external, mimo
|   |   Sensors: mimo_locked, ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   
|   |   |   Freq range: -50.000 to 50.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 1
|   |   |   
|   |   |   Freq range: -50.000 to 50.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |   ID: SBX v5 (0x0069)
|   |   |   Serial: 3114C98
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: 0
|   |   |   |   Name: SBX/CBX RX
|   |   |   |   Antennas: TX/RX, RX2, CAL
|   |   |   |   Sensors: lo_locked
|   |   |   |   Freq range: 400.000 to 4400.000 MHz
|   |   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   |   Bandwidth range: 40000000.0 to 40000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: ads62p44
|   |   |   |   Gain range digital: 0.0 to 6.0 step 0.5 dB
|   |   |   |   Gain range fine: 0.0 to 0.5 step 0.1 dB
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   
|   |   |   Freq range: -200.000 to 200.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: A
|   |   |   ID: SBX v5 (0x0068)
|   |   |   Serial: 3114C98
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: 0
|   |   |   |   Name: SBX/CBX TX
|   |   |   |   Antennas: TX/RX, CAL
|   |   |   |   Sensors: lo_locked
|   |   |   |   Freq range: 400.000 to 4400.000 MHz
|   |   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   |   Bandwidth range: 40000000.0 to 40000000.0 step 0.0 Hz
|   |   |   |   Connection Type: QI
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: ad9777
|   |   |   |   Gain Elements: None
|     _____________________________________________________
|    /
|   |       Mboard: N200r4
|   |   hardware: 2576
|   |   mac-addr: 00:80:2f:0a:de:2d
|   |   ip-addr: 192.168.10.3
|   |   subnet: 255.255.255.255
|   |   gateway: 255.255.255.255
|   |   gpsdo: none
|   |   serial: F42287
|   |   FW Version: 12.4
|   |   FPGA Version: 11.1
|   |   
|   |   Time sources:  none, external, _external_, mimo
|   |   Clock sources: internal, external, mimo
|   |   Sensors: mimo_locked, ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   
|   |   |   Freq range: -50.000 to 50.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 1
|   |   |   
|   |   |   Freq range: -50.000 to 50.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |   ID: SBX (0x0054)
|   |   |   Serial: F5FC43
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: 0
|   |   |   |   Name: SBXv3 RX
|   |   |   |   Antennas: TX/RX, RX2, CAL
|   |   |   |   Sensors: lo_locked
|   |   |   |   Freq range: 400.000 to 4400.000 MHz
|   |   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   |   Bandwidth range: 40000000.0 to 40000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: ads62p44
|   |   |   |   Gain range digital: 0.0 to 6.0 step 0.5 dB
|   |   |   |   Gain range fine: 0.0 to 0.5 step 0.1 dB
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   
|   |   |   Freq range: -200.000 to 200.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: A
|   |   |   ID: SBX (0x0055)
|   |   |   Serial: F5FC43
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: 0
|   |   |   |   Name: SBXv3 TX
|   |   |   |   Antennas: TX/RX, CAL
|   |   |   |   Sensors: lo_locked
|   |   |   |   Freq range: 400.000 to 4400.000 MHz
|   |   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   |   Bandwidth range: 40000000.0 to 40000000.0 step 0.0 Hz
|   |   |   |   Connection Type: QI
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: ad9777
|   |   |   |   Gain Elements: None
```
[out.txt](https://github.com/EttusResearch/uhd/files/4295356/out.txt)

michael-west, 2020-10-12:
@cjosephson Thank you for reporting this and providing the detail.  We have opened an internal issue for further investigation.

mbr0wn, 2021-09-17:
This doesn't happen with other USRPs (e.g, B200 + X410).

closed 2025-05-28 by mbr0wn