EttusResearch/uhd #334: python uhd recv_num_samps() exception after 250 calls

ianccole, 2020-03-13:
## Issue Description ##
Using uhd module in python, recv_num_samps() fails on the 251st iteration throwing an exception.

This is repeatable. It always fails on the 251st attempt.

```
class capture(uhd.usrp.MultiUSRP):
    def __init__(self, args=""):
        super(capture, self).__init__(args)
        subdev = uhd.usrp.SubdevSpec('A:0')
        super(capture, self).set_rx_subdev_spec(subdev)
        super(capture, self).set_rx_antenna('TX/RX')

 dev = capture()
 for ii in range(1,1000):
     samps = dev.recv_num_samps(100000, 4500e6, rate=100e6, gain=38)
     print samps.size

```
Initialised here:
```

```

## Setup Details ##

UHD version origin/UHD-3.15.LTS
USRP X310 

## Expected Behavior ##

To run to the end of the loop
## Actual Behaviour ##

244 100000
245 100000
ERROR_CODE_LATE_COMMAND
ERROR_CODE_LATE_COMMAND
ERROR_CODE_LATE_COMMAND
246 100000
247 100000
248 100000
249 100000
250 100000
Traceback (most recent call last):
  File "/home/pfr347/work/projects/antenna2/usrp_cap.py", line 182, in <module>
    main()
  File "/home/pfr347/work/projects/antenna2/usrp_cap.py", line 163, in main
    samps = dev.recv_num_samps(100000, 4500e6, rate=100e6, gain=38)
  File "/usr/local/lib/python2.7/dist-packages/uhd/usrp.py", line 35, in recv_num_samps
    super(MultiUSRP, self).set_rx_rate(rate, chan)
RuntimeError: EnvironmentError: IOError: [0/DDC_0] sr_write() failed: EnvironmentError: IOError: Block ctrl (CE_03_Port_60) no response packet - AssertionError: bool(buff)
  in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0; uint64_t = long unsigned int]
  at /home/pfr347/work/uhd/host/lib/rfnoc/ctrl_iface.cpp:151

[ERROR] [UHD] Exception caught in safe-call.
  in ctrl_iface_impl<_endianness>::~ctrl_iface_impl() [with uhd::endianness_t _endianness = (uhd::endianness_t)0]
  at /home/pfr347/work/uhd/host/lib/rfnoc/ctrl_iface.cpp:52
this->send_cmd_pkt(0, 0, true); -> EnvironmentError: IOError: Block ctrl (CE_00_Port_31) no response packet - AssertionError: bool(buff)
  in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0; uint64_t = long unsigned int]
  at /home/pfr347/work/uhd/host/lib/rfnoc/ctrl_iface.cpp:151

[ERROR] [UHD] Exception caught in safe-call.
  in ctrl_iface_impl<_endianness>::~ctrl_iface_impl() [with uhd::endianness_t _endianness = (uhd::endianness_t)0]
  at /home/pfr347/work/uhd/host/lib/rfnoc/ctrl_iface.cpp:52
this->send_cmd_pkt(0, 0, true); -> EnvironmentError: IOError: Block ctrl (CE_00_Port_30) no response packet - AssertionError: bool(buff)
  in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0; uint64_t = long unsigned int]
  at /home/pfr347/work/uhd/host/lib/rfnoc/ctrl_iface.cpp:151

[ERROR] [UHD] Exception caught in safe-call.
  in virtual x300_dac_ctrl_impl::~x300_dac_ctrl_impl()
  at /home/pfr347/work/uhd/host/lib/usrp/x300/x300_dac_ctrl.cpp:59
_iface->write_spi(_slaveno, spi_config_t::EDGE_RISE, ((0x01) << 8) | (0xEF), 16); _iface->write_spi(_slaveno, spi_config_t::EDGE_RISE, ((0x02) << 8) | (0x1F), 16); -> EnvironmentError: IOError: Block ctrl (CE_01_Port_40) no response packet - AssertionError: bool(buff)
  in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0; uint64_t = long unsigned int]
  at /home/pfr347/work/uhd/host/lib/rfnoc/ctrl_iface.cpp:151

[ERROR] [UHD] Exception caught in safe-call.
  in virtual x300_adc_ctrl_impl::~x300_adc_ctrl_impl()
  at /home/pfr347/work/uhd/host/lib/usrp/x300/x300_adc_ctrl.cpp:123
this->send_ads62p48_reg(0x40); -> EnvironmentError: IOError: Block ctrl (CE_01_Port_40) no response packet - AssertionError: bool(buff)
  in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0; uint64_t = long unsigned int]
  at /home/pfr347/work/uhd/host/lib/rfnoc/ctrl_iface.cpp:151

[ERROR] [UHD] Exception caught in safe-call.
  in virtual uhd::rfnoc::block_ctrl_base::~block_ctrl_base()
  at /home/pfr347/work/uhd/host/lib/rfnoc/block_ctrl_base.cpp:164
if (get_ctrl_ports().size() > 0) { _start_drain(get_ctrl_ports().front()); } _tree->remove(_root_path); -> EnvironmentError: IOError: [0/Radio_0] sr_write() failed: EnvironmentError: IOError: Block ctrl (CE_01_Port_40) no response packet - AssertionError: bool(buff)
  in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0; uint64_t = long unsigned int]
  at /home/pfr347/work/uhd/host/lib/rfnoc/ctrl_iface.cpp:151

## Steps to reproduce the problem ##

## Additional Information ##

ianccole, 2020-03-13:
Searching reveals a similar issue (iteration number 253) - perhaps unresolved. streamer construct/destruct issue perhaps?

http://lists.ettus.com/pipermail/usrp-users_lists.ettus.com/2018-April/056161.html

ianccole, 2020-03-13:
Here is UHD logging:

https://pastebin.com/Uy3iiaVt

ianccole, 2020-03-16:
I suspect this issue is related
#321

atrnati, 2020-09-24:
Are you able to try reproducing this issue with UHD 4.0.0? We believe this is addressed in the newest release.

michael-west, 2020-10-12:
This is a known issue on UHD 3.15.  It has to do with SID allocation when destroying and recreating streamer objects.  The issue can be mitigated by avoiding destruction and recreation of streamer objects.  They call to recv_num_samps() will create a streamer and destroy it, so a refactoring of the code is recommended in order to persist the streamer.

The issue is resolved on UHD 4.0.0.0 with the restructuring of the architecture.  Unfortunately, it is not something that can be backported to 3.15 so it will not be fixed on 3.15.

closed 2020-10-12 by michael-west