EttusResearch/uhd #336: Build freezes for a long time (94 minutes!) when compiling on macOS

beaugunderson, 2020-03-28:
I acknowledge that I checked the user list, the issues in this repo, the knowledge base, and the manual before opening this issue. I searched for things like slow build, compile freeze, slow compile, compile minutes, compile hours, etc.

## Issue Description ##

I recently upgraded my machine to the latest macOS and Xcode and noticed that while `uhd` does compile, it takes an inordinate amount of time--much longer than I remembered it taking in the past. The vast majority of the time is spent compiling one file: `host/lib/usrp/dboard/twinrx/twinrx_gain_tables.cpp`.

On my last build compiling this file took 94 minutes at 100% CPU usage:

[image]

[image]

The process tree looked like this for the duration:

```
-+= 36576 root sudo port -d install uhd
 \-+- 36578 macports /opt/local/libexec/macports/bin/tclsh8.5 /opt/local/bin/port -d install uhd
   \-+- 37033 macports sh -c  cd "/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build" && /usr/bin/make -j16 -w all VERBOSE=ON
     \-+- 37034 macports /Library/Developer/CommandLineTools/usr/bin/make -j16 -w all VERBOSE=ON
       \-+- 37037 macports /Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/Makefile2 all
         \-+- 37308 macports /Library/Developer/CommandLineTools/usr/bin/make -f lib/CMakeFiles/uhd.dir/build.make lib/CMakeFiles/uhd.dir/build
           \-+- 38037 macports /bin/sh -c cd /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build/lib && /usr/bin/clang++  -DBOOST_ASIO_DISABLE_STD_EXPERIMENTAL_STRING_VIEW -DBOOST_ASIO_DISABLE_STD_STRING_VIEW -DBOOST_ERROR_CODE_HEADER_ONLY -DHAVE_CONFIG_H -DUHD_DLL_EXPORTS -DUHD_LOG_CONSOLE_COLOR -DUHD_LOG_CONSOLE_LEVEL=2 -DUHD_LOG_FILE_LEVEL=2 -DUHD_LOG_MIN_LEVEL=1 -DUHD_RFNOC_ENABLED -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build/include -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/include -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/include -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build/lib/ic_reg_maps -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/convert -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build/lib/convert -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/rfnoc/nocscript -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build/lib/rfnoc/nocscript -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/usrp -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/usrp/common/ad9361_driver -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/usrp/common -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build/lib/transport/nirio/lvbitx -I/opt/local/include/libusb-1.0 -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/deps/rpclib/include  -pipe -Os -stdlib=libc++ -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -fvisibility=hidden -fvisibility-inlines-hidden -DNDEBUG -arch x86_64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -mmacosx-version-min=10.15 -fPIC   -Wall -Wextra -Wsign-compare -std=c++14 -o CMakeFiles/uhd.dir/usrp/dboard/twinrx/twinrx_gain_tables.cpp.o -c /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/usrp/dboard/twinrx/twinrx_gain_tables.cpp
             \-+- 38038 macports /Library/Developer/CommandLineTools/usr/bin/clang++ -DBOOST_ASIO_DISABLE_STD_EXPERIMENTAL_STRING_VIEW -DBOOST_ASIO_DISABLE_STD_STRING_VIEW -DBOOST_ERROR_CODE_HEADER_ONLY -DHAVE_CONFIG_H -DUHD_DLL_EXPORTS -DUHD_LOG_CONSOLE_COLOR -DUHD_LOG_CONSOLE_LEVEL=2 -DUHD_LOG_FILE_LEVEL=2 -DUHD_LOG_MIN_LEVEL=1 -DUHD_RFNOC_ENABLED -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build/include -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/include -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/include -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build/lib/ic_reg_maps -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/convert -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build/lib/convert -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/rfnoc/nocscript -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build/lib/rfnoc/nocscript -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/usrp -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/usrp/common/ad9361_driver -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/usrp/common -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build/lib/transport/nirio/lvbitx -I/opt/local/include/libusb-1.0 -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/deps/rpclib/include -pipe -Os -stdlib=libc++ -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -fvisibility=hidden -fvisibility-inlines-hidden -DNDEBUG -arch x86_64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -mmacosx-version-min=10.15 -fPIC -Wall -Wextra -Wsign-compare -std=c++14 -o CMakeFiles/uhd.dir/usrp/dboard/twinrx/twinrx_gain_tables.cpp.o -c /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/uhd-3.15.0.0/host/lib/usrp/dboard/twinrx/twinrx_gain_tables.cpp
               \--- 38039 macports /Library/Developer/CommandLineTools/usr/bin/clang -cc1 -triple x86_64-apple-macosx10.15.0 -Wdeprecated-objc-isa-usage -Werror=deprecated-objc-isa-usage -emit-obj -disable-free -disable-llvm-verifier -discard-value-names -main-file-name twinrx_gain_tables.cpp -mrelocation-model pic -pic-level 2 -mthread-model posix -mframe-pointer=all -fno-strict-return -masm-verbose -munwind-tables -target-sdk-version=10.15.4 -target-cpu penryn -dwarf-column-info -debugger-tuning=lldb -target-linker-version 556.5 -coverage-notes-file /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_uhd/uhd/work/build/lib/CMakeFiles/uhd.dir/usrp/dboard/twinr [... cut ...]
Compilation succeeds but takes more than an hour to compile one file.

## Steps to reproduce the problem ##

I believe compiling with MacPorts or `cmake` on a version of macOS >= 10.15.1 and Xcode >= 11.3.0 is enough to reproduce this issue.

## Additional Information ##

The reason I care about the compilation time is not just out of convenience--CI systems have strict build time limits, and in the case of Circle CI limits on the duration of time with no output as well as a per-minute charge for build time.

beaugunderson, 2020-03-28:
Here's a sample from `clang`:

[Sample of clang.txt](https://github.com/EttusResearch/uhd/files/4397866/Sample.of.clang.txt)

[image]

beaugunderson, 2020-03-28:
I should acknowledge that this is probably caused by something that broke in a recent version of `clang++` and not necessarily specific to `uhd`, but inevitably people will start to hit this as they upgrade. :)

beaugunderson, 2020-03-28:
Narrowed this down to a single `clang` option: `-Os`. It appears that anything above `-O1` results in the large compile duration.

michaelld, 2020-04-05:
Sorry for the delay; I had to go retrieve my OSX build system for testing & that took a bit to get going again. I've confirmed this issue using the Clang 9.0 compiler; I'm testing 8.0 right now, and it looks to have the same issue; 7.0 does not exhibit the behaviour, so it's a good backup option from Apple Clang. I'll do some testing on Apple Clang too to see what works and what doesn't, mostly for a better understanding of the situation. I'll likely drop the optimization to `-O1` for just this file to get around this issue for the moment. Hopefully it'll be fixed in Clang 10.0. This issue has nothing to do with the OS in my testing; just the compiler.

beaugunderson, 2020-04-05:
Thanks for verifying! Dropping to `-O1` for that file would definitely solve my problem. I lack the cmake knowledge to know how feasible changing that flag based on the version of Clang being used is, but that seems like it might be preferable for users with versions that aren't broken, if there are actual performance differences between `-O1` and `-Os`...

michaelld, 2020-04-07:
Can you test out this patch to verify that it works for you?

[patch-twinrx_gain_tables.cpp-fix-optim.diff.txt](https://github.com/EttusResearch/uhd/files/4445863/patch-twinrx_gain_tables.cpp-fix-optim.diff.txt)

michaelld, 2020-04-14:
I pushed this fix into MacPorts just now: https://github.com/macports/macports-ports/commit/91e45e5b9ba587427e1b3605ab164c506367cc5d

michaelld, 2020-04-14:
It's not ideal, but it works and (1) gets UHD compiling in a reasonable amount of time within that build system; (2) it does no harm ... just reduces optimization from `-Os` to `-O1` ... which I highly doubt will affect runtime noticeably for any user.

The above said, we still need a solution for UHD OSX users who build from source themselves, or using Brew or Fink or whatever other package managers. We're working on this ...

mbr0wn, 2020-09-16:
Fixed by a518c5c771aa9b2e5a537d44b949b84e2d05fcd3

closed 2020-09-16 by mbr0wn