EttusResearch/uhd #337: Ubuntu PPA builds failing for 3.15.0.0 on trusty, bionic

msherman64, 2020-03-31:
## Issue Description ##
The ubuntu PPA is not currently providing 3.15 for versions 14.04 and 18.04 (trusty and bionic)
This page: https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd/+packages shows some builds failing

Please delete this if this is the wrong place to report, but I didn't see any content on the launchpad bugs page.

## Setup Details ##
These link to the failing builds:
https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd/+build/18503413
https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd/+build/18503414
https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd/+build/18503436
https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd/+build/18503437

## Expected Behavior ##
Current version to be provided for all OSs

## Actual Behaviour ##
Build at: https://launchpadlibrarian.net/458657974/buildlog_ubuntu-bionic-amd64.uhd_3.15.0.0-0ubuntu1~bionic1_BUILDING.txt.gz
Fails with this error:
-- Python checking for Python version 3.5 or greater
-- Python checking for Python version 3.5 or greater - found
-- 
-- Python checking for Mako templates 0.4.2 or greater
-- Python checking for Mako templates 0.4.2 or greater - "import mako" failed
-- 
-- Python checking for requests 2.0 or greater
-- Python checking for requests 2.0 or greater - "import requests" failed
-- 
-- Python checking for numpy 1.7 or greater
-- Python checking for numpy 1.7 or greater - "import numpy" failed

mbr0wn, 2020-04-14:
@msherman64 This is a good place to report, thanks for doing that.

michael-west, 2020-10-11:
Trusty (14.04) has a version of cmake that is lower than the required minimum for UHD 3.15.0.0, so that package has been deleted.  All other supported Ubuntu versions (xenial, bionic, and focal) have been updated with Python3 dependencies and all builds have succeeded.  See https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd/+packages

closed 2020-10-11 by michael-west