EttusResearch/uhd #338: set the rpath for everyone

ZeroChaos-, 2020-04-10:
One of gentoo's new QA warnings is unhappy because rpath is set wrong
```
* QA Notice: The following files contain insecure RUNPATHs
 *  Please file a bug about this at https://bugs.gentoo.org/
 *  with the maintainer of the package.
 *   /var/tmp/portage/net-wireless/uhd-3.15.0.0/image/usr/lib64/python2.7/site-packages/uhd/libpyuhd.so
 *     RPATH: /var/tmp/portage/net-wireless/uhd-3.15.0.0/work/uhd-3.15.0.0_build/lib
```
Looking at CmakeLists.txt it seems this is all set for Apple but no one else?  Maybe set it for everyone?

ZeroChaos-, 2020-04-10:
commenting out the If(APPLE) and associated endif makes the gentoo qa check pass

mbr0wn, 2020-05-01:
Took me along time to understand. The RPATH is set incorrectly for the Python API, not the general UHD API. I have a fix in the pipeline.

closed 2020-05-07 by atrnati

mbr0wn, 2020-05-07:
75b7991378e3f465f9d91cba597d98ee5b9054ed