EttusResearch/uhd #339: Low rate of uhd in latest version for B210 devices in windows

FONQRI, 2020-04-11:
## Issue Description ##
I used gnuradio and gr-uhd c++ API for capturing and recording data in windows with B210 and with rfnoc-devel branch that i used i could achieve about 40 mega sample of complex<float> over USB 3.0 and 50 mega sample over USB 3.1 without data lost. but when i used version UHD-3.15.LTS it decreased into 20 MS over USB 3.0 and 25 MS over USB 3.1.

## Setup Details ##

## Expected Behavior ##

UHD-3.15.LTS in windows become same as before in capturing data without data lost(40MS over USB3.0 and 50MS over USB3.1)

## Actual Behaviour ##

It gives me rate about 20MS over USB3.0 and 25MS over USB3.1.

## Steps to reproduce the problem ##

Try capture data with gr-uhd in c++ API on memory and increase samplerate to 40 over USB 3.0

michael-west, 2020-10-11:
@FONQRI Please try setting the otw (over-the-wire) type to sc12 and increasing the `num_recv_frames` to 256 and let us know if that resolves the issue.

mbr0wn, 2021-09-23:
Closing this due to lack of activity.

closed 2021-09-23 by mbr0wn