EttusResearch/uhd #342: Error building PACKAGE

MarkHHansen, 2020-05-07:
## Issue Description ##
I have an issue, when building the PACKAGE project in visual studio on windows. The problem seems to be in cmake_install.cmake. It says an absolute path destination is forbidden. 

## Setup Details ##
Windows 10
Visual Studio 17 v.15.9.22
UHD version: 4.0.0.0-583-g3b59529e
Libusb-1.0.23 (Path C:/Users/myuser/)
Boost_1_68:0 14.1 (Path C:/local)
Python 3.8
CMake 3.17.1

The entries for building the project in CMake are set according to the ettus guide for building windows. 
BOOST_ALL_DYN_LINK: Checked

## Expected Behavior ##
The build works as inteded. 

## Actual Behaviour ##

**When the PACKAGE project is building, the following error occurs:**
3>CPack: -   Install component: pythonapi
3>CMake Error at C:/Users/mhhn/uhd-master/uhd/host/build/python/cmake_install.cmake:42 (message):
3>  ABSOLUTE path INSTALL DESTINATION forbidden (by caller):
3>  C:/local/uhd_install/Lib/site-packages/uhd
3>Call Stack (most recent call first):
3>  C:/Users/mhhn/uhd-master/uhd/host/build/cmake_install.cmake:76 (include)
3>
3>
3>EXEC : CPack error : Error when generating package: UHD
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: The command "setlocal
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: cd C:\Users\mhhn\uhd-master\uhd\host\build
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: if %errorlevel% neq 0 goto :cmEnd
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: C:
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: if %errorlevel% neq 0 goto :cmEnd
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: "C:\Program Files\CMake\bin\cpack.exe" -C Release --config ./CPackConfig.cmake
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: if %errorlevel% neq 0 goto :cmEnd
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: :cmEnd
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: endlocal & call :cmErrorLevel %errorlevel% & goto :cmDone
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: :cmErrorLevel
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: exit /b %1
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: :cmDone
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: if %errorlevel% neq 0 goto :VCEnd
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(138,5): error MSB3073: :VCEnd" exited with code 1.
3>Done building project "PACKAGE.vcxproj" -- FAILED.

## Steps to reproduce the problem ##
The steps to reproduce this error, is just to have the same setup as shown above, and follow this guide:
https://files.ettus.com/manual/page_build_guide.html

There are two warnings when configuring in CMake, and 1 error when build the PACKAGE project. 

## Additional Information ##
CMake output:

Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18363.

Configuring the Python interpreter...
Manually determining build Python version...
Python interpreter: C:/Users/mhhn/AppData/Local/Programs/Python/Python38/python.exe Version: 3.8.2
Override with: -DPYTHON_EXECUTABLE=<path-to-python>
Manually determining runtime Python version...
Python runtime interpreter: C:/Users/mhhn/AppData/Local/Programs/Python/Python38/python.exe Version: 3.8.2
Override with: -DRUNTIME_PYTHON_EXECUTABLE=<path-to-python>
Finding Python Libraries...
Python Libraries: C:/Users/mhhn/AppData/Local/Programs/Python/Python38/libs/python38.lib
Python include directories: C:/Users/mhhn/AppData/Local/Programs/Python/Python38/include
Operating on master branch.
Using UHD Images Directory: C:\local\uhd_install\share\uhd\images

[... 314 lines cut ...]
  * N300
  * N320
  * E320
  * E300
  * OctoClock

######################################################
# UHD disabled components                             
######################################################
  * LibUHD - C API
  * LIBERIO
  * DPDK
  * Manual
  * API/Doxygen
  * Man Pages

******************************************************
* You are building the UHD development master branch.
* For production code, we recommend our stable,
* releases or using the release branch (maint).
******************************************************
Building version: 4.0.0.0-583-g3b59529e
Using install prefix: C:/local/uhd_install
Configuring done
Generating done

atrnati, 2020-09-24:
Hi, thanks for reporting this issue. I believe this has been addressed since your report by commit c68977cf. If you're in a position to give it a try, please let us know if that solves your issue. Thanks again!

michael-west, 2020-10-06:
@MarkHHansen We are closing this as resolved.  Please re-open if it is still an issue for you.

closed 2020-10-06 by michael-west