EttusResearch/uhd #344: Unable to import uhd using Python API for Windows when building from source

Jogans, 2020-05-11:
## Issue Description ##
Hi,
I'm trying to build and install UHD with Python API from source on Windows 10. I've been following  [this guide](https://files.ettus.com/manual/page_build_guide.html) and I'm able to build both the BUILD_ALL and INSTALL projects, but after installing the setup.py file I get an error when I try to import uhd in Python.

## Setup Details ##
My setup is the following:
Windows 10
Visual Studio 2017
MSVC 14.24.28127
CMake 3.17.2
Boost 1.67.0
LibUSB 1.0.22
Python 3.8.2
Mako 1.1.2
NSIS 3.05

## Expected Behavior ##
Being able to import uhd in Python without errors.

## Actual Behaviour ##
This is the output when I install UHD from the setup.py and try to use the module:

```
PS C:\uhd\host\build\python> python .\setup.py install                                                                  running install
running bdist_egg
running egg_info
writing C:\uhd\host\build\python\uhd.egg-info\PKG-INFO
writing dependency_links to C:\uhd\host\build\python\uhd.egg-info\dependency_links.txt
writing requirements to C:\uhd\host\build\python\uhd.egg-info\requires.txt
writing top-level names to C:\uhd\host\build\python\uhd.egg-info\top_level.txt
reading manifest file 'C:\uhd\host\build\python\uhd.egg-info\SOURCES.txt'
writing manifest file 'C:\uhd\host\build\python\uhd.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
copying C:\uhd\host\build\python\uhd\libpyuhd.pyd -> build\lib\uhd
creating build\bdist.win-amd64
creating build\bdist.win-amd64\egg
creating build\bdist.win-amd64\egg\uhd
copying build\lib\uhd\filters.py -> build\bdist.win-amd64\egg\uhd
copying build\lib\uhd\libpyuhd.pyd -> build\bdist.win-amd64\egg\uhd
copying build\lib\uhd\rfnoc.py -> build\bdist.win-amd64\egg\uhd
copying build\lib\uhd\types.py -> build\bdist.win-amd64\egg\uhd
copying build\lib\uhd\__init__.py -> build\bdist.win-amd64\egg\uhd
byte-compiling build\bdist.win-amd64\egg\uhd\filters.py to filters.cpython-38.pyc
byte-compiling build\bdist.win-amd64\egg\uhd\rfnoc.py to rfnoc.cpython-38.pyc
byte-compiling build\bdist.win-amd64\egg\uhd\types.py to types.cpython-38.pyc
byte-compiling build\bdist.win-amd64\egg\uhd\__init__.py to __init__.cpython-38.pyc
creating build\bdist.win-amd64\egg\EGG-INFO
copying C:\uhd\host\build\python\uhd.egg-info\PKG-INFO -> build\bdist.win-amd64\egg\EGG-INFO
copying C:\uhd\host\build\python\uhd.egg-info\SOURCES.txt -> build\bdist.win-amd64\egg\EGG-INFO
copying C:\uhd\host\build\python\uhd.egg-info\dependency_links.txt -> build\bdist.win-amd64\egg\EGG-INFO
copying C:\uhd\host\build\python\uhd.egg-info\not-zip-safe -> build\bdist.win-amd64\egg\EGG-INFO
copying C:\uhd\host\build\python\uhd.egg-info\requires.txt -> build\bdist.win-amd64\egg\EGG-INFO
copying C:\uhd\host\build\python\uhd.egg-info\top_level.txt -> build\bdist.win-amd64\egg\EGG-INFO
writing build\bdist.win-amd64\egg\EGG-INFO\native_libs.txt
creating dist
creating 'dist\uhd-4.0.0-py3.8.egg' and adding 'build\bdist.win-amd64\egg' to it
removing 'build\bdist.win-amd64\egg' (and everything under it)
Processing uhd-4.0.0-py3.8.egg
creating c:\users\radm\appdata\local\programs\python\python38\lib\site-packages\uhd-4.0.0-py3.8.egg
Extracting uhd-4.0.0-py3.8.egg to c:\users\radm\appdata\local\programs\python\python38\lib\site-packages
Adding uhd 4.0.0 to easy-install.pth file

Installed c:\users\radm\appdata\local\programs\python\python38\lib\site-packages\uhd-4.0.0-py3.8.egg
Processing dependencies for uhd==4.0.0
Searching for numpy==1.18.4
Best match: numpy 1.18.4
Adding numpy 1.18.4 to easy-install.pth file
Installing f2py-script.py script to C:\Users\radm\AppData\Local\Programs\Python\Python38\Scripts
Installing f2py.exe script to C:\Users\radm\AppData\Local\Programs\Python\Python38\Scripts

Using c:\users\radm\appdata\local\programs\python\python38\lib\site-packages
Finished processing dependencies for uhd==4.0.0
PS C:\uhd\host\build\python> python                                                                                     Python 3.8.2 (tags/v3.8.2:7b3ab59, Feb 25 2020, 23:03:10) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import uhd
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\uhd\host\build\python\uhd\__init__.py", line 10, in <module>
    from . import types
  File "C:\uhd\host\build\python\uhd\types.py", line 10, in <module>
    from . import libpyuhd as lib
ImportError: DLL load failed while importing libpyuhd: The specified module could not be found.
```

## Steps to reproduce the problem ##
Build UHD from source following [this](https://files.ettus.com/manual/page_build_guide.html) guide for Windows and enable Python API in CMake. 
After building ALL_BUILD and INSTALL, open PowerShell and run the following code
`CD C:\uhd\host\build\python (or wherever the UHD code is checked out)
python .\setup.py install
python
import uhd
`

## Additional Information ##
I get two warnings when configuring in CMake, but they do not seem to be related to the issue regarding importing uhd in Python. These two errors are the following:

`CMake Warning (dev) at C:/Program Files (x86)/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (USB1).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.17/Modules/FindPkgConfig.cmake:45 (find_package_handle_standard_args)
  cmake/Modules/FindUSB1.cmake:6 (include)
  lib/CMakeLists.txt:62 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/Program Files (x86)/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (LIBUSB)
  does not match the name of the calling package (USB1).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Modules/FindUSB1.cmake:61 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  lib/CMakeLists.txt:62 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.`

atrnati, 2020-09-25:
I think you may be running into an issue with a change that was made to the [Windows version of Python to more securely resolve paths for locating modules](https://docs.python.org/3/whatsnew/3.8.html#bpo-36085-whatsnew). We documented a way to address this in the [Python API section of the UHD manual](https://files.ettus.com/manual/page_python.html) (although we only discovered this recently ourselves before the UHD 4.0.0 release, so it hasn't been documented for long!). 

Can you give the suggestion there a try and see if you are able to resolve your issue?

michael-west, 2020-10-12:
Assuming this is resolved with the method documented in the manual.  Please re-open if it is not.  Thank you!

closed 2020-10-12 by michael-west