EttusResearch/uhd #345: NMEA strings not implemented for USRP N310

lunar-mosiur, 2020-05-29:
## Issue Description ##

I am facing problems to run the UHD's default C++ code query_gpsdo_sensors.cpp. When I run it it generates the following messages:

...
...
...
Trying to align the device time to GPS time...
GPS and UHD Device time are aligned.
last_pps: 1590057076 vs gps: 1590057076.
NMEA strings not implemented for this device.
GPS Epoch time at last PPS: 1590057077.00000 seconds
UHD Device time last PPS:   1590057077.00000 seconds
UHD Device time right now:  1590057077.82204 seconds
PC Clock time:              1590057077 seconds

Done!

[ERROR] [UHD] Exception caught in safe-call.
  in ctrl_iface_impl<_endianness>::~ctrl_iface_impl() [with uhd::endianness_t _endianness = (uhd::endianness_t)0]
  at /home/ubuntu/Softwares/GNURadio/workarea-uhd/uhd/host/lib/rfnoc/ctrl_iface.cpp:52
this->send_cmd_pkt(0, 0, true); -> EnvironmentError: IOError: Block ctrl (CE_03_Port_61) packet parse error - EnvironmentError: IOError: Expected packet index: 23 Received index: 27
[ERROR] [UHD] Exception caught in safe-call.
  in ctrl_iface_impl<_endianness>::~ctrl_iface_impl() [with uhd::endianness_t _endianness = (uhd::endianness_t)0]
  at /home/ubuntu/Softwares/GNURadio/workarea-uhd/uhd/host/lib/rfnoc/ctrl_iface.cpp:52
this->send_cmd_pkt(0, 0, true); -> EnvironmentError: IOError: Block ctrl (CE_03_Port_60) packet parse error - EnvironmentError: IOError: Expected packet index: 42 Received index: 46
## Setup Details ##

## Expected Behavior ##

NMEA strings containing latitude, longitude, altitude from the GPS.
## Actual Behaviour ##

NMEA strings not implemented for this device.
## Steps to reproduce the problem ##

Compile query_gpsdo_sensors.cpp for Ubuntu 18.04 and UHD v3.15.0.0 (UHD_3.15.0.HEAD-0-gaea0e2de.). Then run. 
## Additional Information ##

The reason is unknown for the error message in the end. Creating another issue for that.

atrnati, 2020-09-24:
Can you see if you are able to reproduce the issue you report with the UHD 4.0.0 release?

michael-west, 2020-10-12:
The embedded devices have a GPSD and not a GPSDO, so they provide a GPGGA string based on the TPV and SKY sensor data but they do not provide a GPRMC string.  This is by design.  Please use `multi_usrp::get_mboard_sensors()` to see a full list of available GPS sensors.

We are in the process of updating the `query_gpsdo_sensors` to print out the GPGGA string for the embedded devices thanks to you reporting this.  Thank you!

closed 2020-10-12 by michael-west