EttusResearch/uhd #347: Build fails with Boost 1.73.0

jwakely, 2020-06-03:
## Issue Description ##

Trying to build with Bosot 1.73.0 results in dozens of failures, due to changes in Boost.Bind. One example:

```
/builddir/build/BUILD/uhd-3.15.0.0-rc2/host/lib/rfnoc/dma_fifo_block_ctrl_impl.cpp: In constructor 'dma_fifo_block_ctrl_impl::dma_fifo_block_ctrl_impl(const uhd::rfnoc::make_args_t&)':
/builddir/build/BUILD/uhd-3.15.0.0-rc2/host/lib/rfnoc/dma_fifo_block_ctrl_impl.cpp:61:21: error: '_1' was not declared in this scope
   61 |                     _1,
      |                     ^~
```

Other headers get this message:

```
In file included from /usr/include/boost/bind.hpp:30,
                 from /builddir/build/BUILD/uhd-3.15.0.0-rc2/host/lib/rfnoc/nocscript/function_table.cpp:10:
/usr/include/boost/bind.hpp:36:1: note: '#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.'
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
```

I've used this patch to get it to build in Fedora:
https://src.fedoraproject.org/rpms/uhd/blob/745534ebabc68ae009475424e8bcee0fc5f0317a/f/uhd-bind-placeholders.patch

mbr0wn, 2020-07-16:
I somehow wasn't able to reproduce the bug, even though it's super obvious. Maybe some CMake / Boost version issue on my end. However, I think a proper usage of includes will fix this.

mbr0wn, 2020-07-16:
Lol immediately after hitting send I realized what I was doing wrong. Yeah I can see the bug. A better fix is to add the appropriate includes everywhere. I notice a *lot* of new, unrelated warnings with Boost 1.73 but one thing at a time.

mbr0wn, 2020-07-22:
Fixedb by 13caaf001

closed 2020-07-22 by mbr0wn

TylerWerne40, 2024-07-10:
@mbr0wn How did you resolve this? We have to use 1.73 for boost and we have to use UHD 3.13 for a project.

mbr0wn, 2024-07-11:
Check out the commit above. If it's not on 3.13, try backporting.

TylerWerne40, 2024-07-11:
> Check out the commit above. If it's not on 3.13, try backporting.

Still trying to get UHD 3.13 working. Adding UHDBoost.cmake didn't address it. Are you suggesting using an older version of boost?