EttusResearch/uhd #351: X310. When turning on TX and RX in one prog, RX sleeping for a minute. To check, run txrx_loopback_to_file  --file usrp_samples.dat --tx-rate 1000000 --rx-rate 1000000 --tx-freq 950000000 --rx-freq 950000000  --tx-ant="TX/RX" --rx-ant="TX/RX" --ref="internal"

medvedvvs, 2020-06-08:
## Issue Description ##

## Setup Details ##

## Expected Behavior ##

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

medvedvvs, 2020-06-08:
X310, PCIe. When turning on TX and RX in one prog, RX is sleeping for a minute. To check, run txrx_loopback_to_file  --file usrp_samples.dat --tx-rate 1000000 --rx-rate 1000000 --tx-freq 950000000 --rx-freq 950000000  --tx-ant="TX/RX" --rx-ant="TX/RX" --ref="internal"

atrnati, 2020-09-24:
Would you be able to provide more details on your setup, including exact version of UHD you're using, system details (OS and version), and USRP hardware configuration (device, daughterboard(s), transports, etc.)?

NI-LAm, 2020-10-02:
Can one do TX and RX on the same antenna port at the same time? Shouldn't the arguments be `--tx-ant="TX/RX --rx-ant="RX2"`? Configured like above I would assume the RX to wait for the TX to finish transmitting which might explain why there is a delay in RX.

michaelld, 2020-10-02:
> Can one do TX and RX on the same antenna port at the same time?

Nope. Can do -either- TX -or- RX.

michaelld, 2020-10-06:
@NI-LAm I stand corrected: I am told that one -can- do -simultaneous- TX and RX on the same TX/RX port (at least on the X310; I'm checking for others). The TX will bleed into the RX and so your app will need to find a way to ignore or filter the VERY LOUD TX'd signal when doing RX. To be honest: I just learned of this use option; I've never tried it myself; I'm a little surprised that this works! That said: I can see some pretty cool use cases for it, with the correct RX filtering out of the TX signal.

michaelld, 2020-10-06:
See for example this [GRCon 2017 presentation](https://www.gnuradio.org/grcon/grcon17/presentations/full_duplex_radio/Adam-Parower-Full-Duplex-Radio.pdf) discussing the topic of how to do self-interference cancellation when using the same port for TX and RX.

michael-west, 2020-10-12:
Yes, you can use the same port simultaneously for TX and RX.  The RF switch for the TX/RX port will be switched to the TX path and the signal on the RX path will be attenuated by the amount of the isolation provided by the switch.

@medvedvvs This could be an issue on the PCIe transport.  Are you able to test using 10 GbE?

michael-west, 2020-10-12:
Redirecting to Support for reproduction and validation.