EttusResearch/uhd #352: More than one instance of constructor "uhd::time_spec_t::time_spec_t" matches the argument list

teku45, 2020-06-23:
## Issue Description ##
Ported UHD recording C++ script from Ubuntu 18.04 to MacOS 10.15. The code compiles and runs fine on Ubuntu but I get the following error on Mac.

**Full Error Readout**
> more than one instance of constructor "uhd::time_spec_t::time_spec_t" matches the argument list: -- function "uhd::time_spec_t::time_spec_t(double secs = (0.0))" -- function "uhd::time_spec_t::time_spec_t(int64_t full_secs, double frac_secs = (0.0))" -- argument types are: (time_t)

## Setup Details ##
Linux machine:
`[INFO] [UHD] linux; GNU C++ version 7.4.0; Boost_106501; UHD_3.14.1.1-release`

Mac machine
` [INFO] [UHD] Mac OS; Clang version 11.0.0 (clang-1100.0.33.17); Boost_107100; UHD_3.15.0.0-MacPorts-Release`

## Expected Behavior ##
Error does not occur

## Actual Behavior ##
Above error occurs at compile time

## Relevant code lines that cause the issue ##

> `uhd::time_spec_t gps_time = uhd::time_spec_t(time_t(usrp->get_mboard_sensor("gps_time", 0).to_int()));`

> `gps_time = uhd::time_spec_t(time_t(usrp->get_mboard_sensor("gps_time", 0).to_int()));`

michaelld, 2020-06-23:
any way you can post your code for us to test?

teku45, 2020-06-23:
Here is the C++ file saved as a .txt since GitHub isn't letting me upload .cpp. works just the same if you change the extension
[XM_time.txt](https://github.com/EttusResearch/uhd/files/4821357/XM_time.txt)

michaelld, 2020-06-23:
OK. I see this too. If you change the lines to first convert to "int64_t" instead of "time_t" your code compiles cleanly, and I believe that's the correct solution since the sensor is returning an integer value of some sort.

teku45, 2020-07-07:
Thank you this sorted the issue out

closed 2020-07-07 by teku45