EttusResearch/uhd #353: Issue with uhd setting up for RFNoC with Docker

SerkanErgunAAU, 2020-06-26:
## Issue Description ##

## Setup Details ##
Using Docker and  ubuntu 16.04 image. I installed Mako, numpy,.. with pip3 to resolve an issue that these packages were not found. Please be aware that some packages are installed multiple times, I excuse for the the untidy code.

## Expected Behavior ##
should build properly as described in https://kb.ettus.com/Getting_Started_with_RFNoC_Development. Please check the Dockerfile's code below to reproduce.

## Actual Behaviour ##
Here is the last part of the terminal output
`Step 7/11 : RUN pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git     && pybombs recipes add ettus git+https://github.com/EttusResearch/ettus-pybombs.git     && pybombs recipes add gr-iio git+https://github.com/analogdevicesinc/gr-iio.git
 ---> Using cache
 ---> e068b6564a05
Step 8/11 : RUN pybombs config -P uhd gitbranch UHD-3.15.LTS
 ---> Using cache
 ---> 7cc425b3bf86
Step 9/11 : RUN pybombs --config makewidth=2 prefix init ~/rfnoc -R rfnoc -a rfnoc
 ---> Running in 57a57350ff27
[INFO] Prefix Python version is: 3.5.2
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Creating directory `/root/rfnoc'
[INFO] Prefix Python version is: 3.5.2
[INFO] Creating new config file /root/rfnoc/.pybombs/config.yml
[INFO] Prefix Python version is: 3.5.2
[INFO] Installing default packages for prefix...
[INFO] 
  - gnuradio
  - uhd
  - gr-ettus
  - pyqt5
  - uhd-fpga
[INFO] Install python-apt to speed up apt processing.
[INFO] Phase 1: Creating install tree and installing binary packages:
[INFO] Executing command with elevated privileges: `sudo -H apt-get -y install python3-pyqt5'
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libevdev2 libinput-bin libinput10 libmtdev1 libqt5clucene5 libqt5core5a
  libqt5dbus5 libqt5designer5 libqt5gui5 libqt5help5 libqt5network5
  libqt5printsupport5 libqt5sql5 libqt5sql5-sqlite libqt5svg5 libqt5test5
  libqt5widgets5 libqt5xml5 libwacom-bin libwacom-common libwacom2
  libxcb-icccm4 libxcb-image0 libxcb-keysyms1 libxcb-render-util0 libxcb-util1
  libxcb-xkb1 libxkbcommon-x11-0 python3-sip qttranslations5-l10n
Suggested packages:
  libqt5libqgtk2 qt5-image-formats-plugins qtwayland5 python3-pyqt5-dbg
The following NEW packages will be installed:
  libevdev2 libinput-bin libinput10 libmtdev1 libqt5clucene5 libqt5core5a
  libqt5dbus5 libqt5designer5 libqt5gui5 libqt5help5 libqt5network5
  libqt5printsupport5 libqt5sql5 libqt5sql5-sqlite libqt5svg5 libqt5test5
  libqt5widgets5 libqt5xml5 libwacom-bin libwacom-common libwacom2
  libxcb-icccm4 libxcb-image0 libxcb-keysyms1 libxcb-render-util0 libxcb-util1
  libxcb-xkb1 libxkbcommon-x11-0 python3-pyqt5 python3-sip
  qttranslations5-l10n
0 upgraded, 31 newly installed, 0 to remove and 1 not upgraded.
Need to get 14.5 MB of archives.
After this operation, 60.4 MB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu xenial/main amd64 libmtdev1 amd64 1.1.5-1ubuntu2 [13.8 kB]
Get:2 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 libqt5core5a amd64 5.5.1+dfsg-16ubuntu7.7 [1817 kB]
Get:3 http://archive.ubuntu.com/ubuntu xenial/main amd64 libevdev2 amd64 1.4.6+dfsg-1 [28.0 kB]
Get:4 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 libwacom-common all 0.22-1~ubuntu16.04.1 [32.0 kB]
Get:5 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 libwacom2 amd64 0.22-1~ubuntu16.04.1 [17.3 kB]
Get:6 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 libinput-bin amd64 1.6.3-1ubuntu1~16.04.1 [9040 B]
Get:7 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 libinput10 amd64 1.6.3-1ubuntu1~16.04.1 [74.0 kB]
Get:8 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 libqt5dbus5 amd64 5.5.1+dfsg-16ubuntu7.7 [175 kB]
Get:9 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 libqt5network5 amd64 5.5.1+dfsg-16ubuntu7.7 [541 kB]
Get:10 http://archive.ubuntu.com/ubuntu xenial/main amd64 libxcb-icccm4 amd64 0.4.1-1ubuntu1 [10.4 kB]
Get:11 http://archive.ubuntu.com/ubuntu xenial/main amd64 libxcb-util1 amd64 0.4.0-0ubuntu3 [11.2 kB]
Get:12 http://archive.ubuntu.com/ubuntu xenial/main amd64 libxcb-image0 amd64 0.4.0-1build1 [12.3 kB]
Get:13 http://archive.ubuntu.com/ubuntu xenial/main amd64 libxcb-keysyms1 amd64 0.4.0-1 [8406 B]
[... 901 lines cut ...]
62. # https://github.com/analogdevicesinc/gr-iio/tree/upgrade-3.8
63. #From here, PyBOMBS can be used to setup and install the development environment/sandbox by running the following command: 
64. RUN pybombs --config makewidth=2 prefix init ~/rfnoc -R rfnoc -a rfnoc
65. 
66. #Then, it is necessary to setup the PyBOMBS environment, so that the system/terminal session 
67. #will have the environmental variables pointing to this newly created prefix, which is done with the following commands: 
68. RUN cd ~/rfnoc \
69.     && source ./setup_env.sh
70. 
71. 
72. #These are the Hash commits you need to use in order to have the correct version running
73. #uhd: ec9138eb6634b0af106762832c7518c887576a94
74. #uhd-fpga: b0890fa97ef3dc7d90ed8047d678ca280c72ad61
75. #gr-ettus: 457dc0a203045aa5c201655dafba4f7b90c951ee
76. RUN cd ~/rfnoc/src/uhd \
77.     && git checkout ec9138eb6634b0af106762832c7518c887576a94 \
78.     && cd ~/rfnoc/src/uhd-fpga\
79.     && git checkout b0890fa97ef3dc7d90ed8047d678ca280c72ad61 \
80.     && cd ~/rfnoc/src/gr-ettus\
81.     && git checkout 457dc0a203045aa5c201655dafba4f7b90c951ee 
82. 
83. # Check USRP connection and load the image to the USRP

`
I hope that you have an solution for this issue.

closed 2020-06-30 by SerkanErgunAAU

sdrhertz, 2020-07-04:
What was the solution?

I have followed the same instructions and come across the same error, using Ubuntu 16 directly though...

SerkanErgunAAU, 2020-07-04:
I specified a Version of UHD (See in "Further" Edit above) before running pybombs prefix .... Although I only reached to building GNU Radio.