EttusResearch/uhd #356: AD9361 Driver Violating Maximums

mcarosino, 2020-07-03:
## Issue Description ##
The AD9361 Driver appears to be violating stated maximum's for the 9361 configuration in certain instances. Consider the following code taken from here:
https://github.com/EttusResearch/uhd/blob/master/host/lib/usrp/common/ad9361_driver/ad9361_device.cpp#L1511

When the requested sample rate is say 55M, for example, the divfactor will be 12 and BBPLL will be tuned to accommodate 12*55=660M (resulting BB PLL is 1.23G). The ADC clk will be 660M and the DAC clk will be 330M. 

The driver notes:
>    /* The DAC clock must be <= 336e6, and is either the ADC clock or 1/2 the
     * ADC clock.*/

However, ADI has stated the maximum ADC clk is actually 640M and the maximum DAC clk is 320M. These limits are enforced in the ad9361 filter wizard as well as the AD9361 linux driver. See the following posts also that confirm. 
https://ez.analog.com/wide-band-rf-transceivers/design-support/f/q-a/163380/ad9361-filter-max-dac-clk-rate
https://ez.analog.com/microcontroller-no-os-drivers/f/q-a/85865/ad9361-sampling-frequency-issue

It looks like 58M might be the worst scenario, this will result in the BBPLL being tuned to accommodate 12*58=696M (corresponding to pll freq of 1.392G) and thus an ADC clk=696M and DAC clk of 348M, well over the maximums.

To be fair, I have't proven this causes a noticeable problem in the performance of the chip with the clocks being overdriven like this, I assume these maximums come from a timing analysis done by ADI. In practice I guess you might be safe to go a bit over although at some point we'd surely start seeing performance degradation.

atrnati, 2020-09-24:
Thank you very much for reporting this! We've created an issue to track this internally.

mbr0wn, 2021-09-23:
@mcarosino Excellent observation. Our driver predates the official ADI driver, and used some older recommendations. We are very hesitant to change it -- it seems to work, and we haven't gotten SI-related or other complaints so far. But you're right, a comment that we are aware of this would be useful. We'll add that.

closed 2021-09-23 by mbr0wn