EttusResearch/uhd #361: usrp probe

MoneyCatcher, 2020-07-21:
## Issue Description ##

**uhd_find_devices** and **uhd_usrp_probe** and args="use_dpdk=1" results in an error.

```
[ERROR] [X300] X300 Network discovery error AssertionError: port_id >= 0
  in uhd::transport::dpdk_simple_impl::dpdk_simple_impl(uhd::transport::uhd_dpdk_ctx&, const string&, const string&, bool)
  at /home/demo/GIT/uhd/host/lib/transport/dpdk_simple.cpp:29
```

## Setup Details ##

Ubuntu 18.04
UHD 3.15
N310

## Expected Behavior ##

**uhd_find_devices**

Don't show     **reachable: No**
instead show **addr= \<IP\>**

## Actual Behaviour ##
--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:

    serial: <number>
    claimed: False
    mgmt_addr: <IP>
    product: n310
    reachable: No
    type: n3xx

## Steps to reproduce the problem ##

`sudo uhd_find_devices --args use_dpdk=1
`
## Additional Information ##
This only affects **uhd_find_device** and **uhd_usrp_probe** a self written app is working

mbr0wn, 2020-09-16:
So the error is thrown by the X300 code. What if you add `type=n300`?

MoneyCatcher, 2020-09-21:
With:

`sudo uhd_find_devices --args use_dpdk=1,type=n3xx`

The error was not shown, but reachable is still No

--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    serial: 316A5B5
    claimed: False
    mgmt_addr: 192.168.110.3
    product: n310
    reachable: No
    type: n3xx

I need to define the argument string like this, so I is there currently no way for usrp_probe to ask the mgmt ip for the data plane ip?

`sudo uhd_find_devices --args use_dpdk=1,type=n3xx,mgmt_addr0=192.168.110.3,addr0=192.168.111.3,second_addr0=192.168.112.3'

--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    serial: 316A5B5
    claimed: False
    mgmt_addr: 192.168.110.3
    product: n310
    type: n3xx

Finally, I'm still thinking, usrp probe should not bring a message like reachable no, when it is reachable!
Maybe refactoring the code a little bit, helps the unskilled user, Even when the message shows an error for the X series instead of N.

Thanks a lot, for your review
BR Matthias

neelpandeya, 2020-09-23:
Which version of DPDK are you using?

Could you report this issue to "support@ettus.com" ?

MoneyCatcher, 2020-09-23:
Currently, I'm using release 3.15.0.0.

mbr0wn, 2021-09-24:
@MoneyCatcher yeah you have to provide both addr and mgmt_addr when using DPDK (see also the manual https://files.ettus.com/manual_archive/v3.15.0.0/html/page_dpdk.html#dpdk_device_args). This is not a priority for us to change. That error message above is annoying, though, and we'll take a look at it.

mbr0wn, 2024-06-21:
Given there is a workaround, and this is reported for UHD 3.15, we won't be fixing this on UHD 3.

closed 2024-06-21 by mbr0wn