EttusResearch/uhd #368: UHD 4.0.0.0-rc1, X310, multi-usrp: runtime error when specifying subdevice

ab-kbg, 2020-08-27:
## Issue Description ##
Specifying subdevice causes program to fail. This hinders dc offset calibration for instance, since only A output gets used by default unless subdevice is set on the command line.

## Setup Details ##
UHD_4.0.0.HEAD-0-g4c8ba577 (should be equivalent to rc1), Fedora 32 Linux
1 X310s motherboard, 192.168.10.2
2 UBX-40 daughterboards
Laptop with IP 192.168.10.1 connected to the X310 via 1Gbit ethernet switch.
OctoClock-G connected to X310, providing 1 PPS and 10 MHz (not required to reproduce)

## Expected Behavior ##
`uhd_cal_tx_dc_offset --verbose --args="addr=192.168.10.2" --subdev="B:0"` should run dc calibration on B TX output.
`tx_waveforms --args="addr=192.168.10.2" --rate 1e6 --freq 350e6 --gain 15 --subdev="B:0"` should generate a sinusoid at the B TX output.

## Actual Behaviour ##
Above mentioned programs fail with output: 
```
$ uhd_cal_tx_dc_offset --verbose --args="addr=192.168.10.2" --subdev="B:0"

Creating the usrp device with: addr=192.168.10.2,ignore_cal_file=1,ignore-cal-file=1...
[INFO] [UHD] linux; GNU C++ version 10.2.1 20200723 (Red Hat 10.2.1-1); Boost_106900; UHD_4.0.0.HEAD-0-g4c8ba577
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
Error: map::at
```
and
```
$ tx_waveforms --args="addr=192.168.10.2" --rate 1e6 --freq 350e6 --gain 15 --subdev="B:0"

Creating the usrp device with: addr=192.168.10.2...
[INFO] [UHD] linux; GNU C++ version 10.2.1 20200723 (Red Hat 10.2.1-1); Boost_106900; UHD_4.0.0.HEAD-0-g4c8ba577
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
Error: map::at
```

Same issue happens when specifying "A:0 B:0" also. When no subdev spec is given, the programs operate with default subdev spec just fine. When a poorly formatted subdev string like "B" is provided, UHD does complain that no matching radio is found.

## Steps to reproduce the problem ##
Run above commands.

## Additional Information ##
Traces attached:
[uhd_cal_tx_dc_offset_trace.log](https://github.com/EttusResearch/uhd/files/5136648/uhd_cal_tx_dc_offset_trace.log)
[tx_waveforms_trace.log](https://github.com/EttusResearch/uhd/files/5136649/tx_waveforms_trace.log)

dwidmann, 2020-09-03:
Slightly different conditions (built from uhd 4.0-RC1 tag, X310, UBX-160 and TwinRX daughterboards, connected via 1GbE, using a relatively simple C++ program loosely based on txrx_loopback_to_file example), but I was able to reproduce this bug.

ab-kbg, 2020-09-15:
With branch UHD-4.0, 90ce6062b6b5df2eddeee723777be85108e4e7c7, this issue is not present.

dwidmann, 2020-09-16:
90ce606 fixes this issue for me also.

michael-west, 2020-09-21:
Closing as resolved.

closed 2020-09-21 by michael-west