EttusResearch/uhd #369: build errors uhd 4.0rc1 and boost 1_67_0

rcarmich, 2020-08-28:
## Issue Description ##
Building UHD 4rc1 with boost 1_67_0 results in

```
boost_1_67_0/include/boost/lockfree/spsc_queue.hpp:140:43: error: ‘next’ is not a member of ‘boost’
```
## Expected Behavior ##
Compilation succeeds

## Actual Behaviour ##
Compilation fails
```
boost_1_67_0/include/boost/lockfree/spsc_queue.hpp:140:43: error: ‘next’ is not a member of ‘boost’
```
## Fix ##
spsc_queue.hpp should include:

```
#include <boost/fusion/iterator/next.hpp>
#include <boost/fusion/include/next.hpp>
```

and fully reference

``boost::fusion::next``

rcarmich, 2020-08-28:
Let me revise this with a likely better suggestion - if you just include <boost/next_prior.hpp> then boost::next works without going through ::fusion.

mbr0wn, 2020-09-07:
@rcarmich Thansk for the feedback. FYI, we'll need a solution that works all the way down to Boost 1.58, so we need to make sure we're not breaking older Boosts which is usually the hold-up for things like this.

mbr0wn, 2020-09-09:
@rcarmich So a couple of odd things. First, you're suggesting to change a file in Boost itself -- was that intentional? Second, `boost::next` is never used in UHD 4.0. Third, we've tested UHD with a bunch of Boost versions, not sure if 1.67 was one of them, but newer versions are. Can you provide more info what you were testing? For example, where did that Boost come from. Is it part of a regular Ubuntu/Fedora/CentOS/Whatever system?

rcarmich, 2020-09-09:
@mbr0wn  you're right, that was my mistake - I shouldn't have suggested that fix, please ignore. But I believe the issue definitely exists.

Here's an easy way to reproduce (excuse the inefficient Dockerfile) - see attached.
[Dockerfile.txt](https://github.com/EttusResearch/uhd/files/5197217/Dockerfile.txt)

mbr0wn, 2020-09-11:
Fix is on the way.

lv254165754, 2021-11-02:
You should see this:
https://github.com/boostorg/lockfree/commit/12726cda009a855073b9bedbdce57b6ce7763da2?diff=split