EttusResearch/uhd #370: v4.0.0.0-rc1 in rx_ascii_art_dft and others --likely on a call to set_rx_freq for N200/UBX

kb1vc, 2020-08-29:
## Issue Description ##
From a build of commit 4c8ba5775286c4b1dee5ccd9e1493862e22d1862  rx_ascii_art_dft fails like this: 
```
$ /usr/local/lib64/uhd/examples/rx_ascii_art_dft --freq=50.056e6 --gain 1000 --rate 1e6 --args='addr=192.168.10.2'

Creating the usrp device with: addr=192.168.10.2...
[INFO] [UHD] linux; GNU C++ version 9.3.1 20200408 (Red Hat 9.3.1-2); Boost_106900; UHD_4.0.0.HEAD-0-g4c8ba577
[INFO] [USRP2] Opening a USRP2/N-Series device...
[INFO] [USRP2] Current recv frame size: 1472 bytes
[INFO] [USRP2] Current send frame size: 1472 bytes
Using Device: Single USRP:
  Device: USRP2 / N-Series Device
  Mboard 0: N200
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: UBX RX
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: UBX TX

Setting RX Rate: 1.000000 Msps...
Actual RX Rate: 1.000000 Msps...

Setting RX Freq: 50.056000 MHz...

```
and then it hangs.  Looking at the source for rx_ascii_art_dft.cpp the next thing that happens after the "Setting RX" message is a call to set_rx_freq. 

This does *not* happen when I run the same program on a B210 like this:
```
$ /usr/local/lib64/uhd/examples/rx_ascii_art_dft --freq=50.056e6 --gain 1000 --rate 1e6 --args='type=b200'
```
SoDaRadio commit 7692d729ba90fe137b2eadca294499d0cb490ad5 from [git@github.com:kb1vc/SoDaRadio.git ](url) hangs on its first call to set_rx_freq as well. But SoDaRadio fails for both the N200/UBX and the B210. 

## Setup Details ##

Fedora 30
gcc version gcc (GCC) 9.3.1 20200408 (Red Hat 9.3.1-2)
[INFO] [UHD] linux; GNU C++ version 9.3.1 20200408 (Red Hat 9.3.1-2); Boost_106900; UHD_4.0.0.HEAD-0-g4c8ba577
uhd_usrp_probe output: 
```
$ uhd_usrp_probe 
[INFO] [UHD] linux; GNU C++ version 9.3.1 20200408 (Red Hat 9.3.1-2); Boost_106900; UHD_4.0.0.HEAD-0-g4c8ba577
[INFO] [USRP2] Opening a USRP2/N-Series device...
[INFO] [USRP2] Current recv frame size: 1472 bytes
[INFO] [USRP2] Current send frame size: 1472 bytes
  _____________________________________________________
 /
|       Device: USRP2 / N-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: N200
|   |   hardware: 2560
|   |   mac-addr: a0:36:fa:26:31:5b
|   |   ip-addr: 192.168.10.2
|   |   subnet: 255.255.255.255
|   |   gateway: 255.255.255.255
|   |   gpsdo: none
|   |   serial: E5R11YBUN
|   |   FW Version: 12.4
|   |   FPGA Version: 11.1
|   |   
|   |   Time sources:  none, external, _external_, mimo
|   |   Clock sources: internal, external, mimo
|   |   Sensors: mimo_locked, ref_locked

```
## Expected Behavior ##

I expected the call to set_rx_freq to return.  It does not.  

## Actual Behaviour ##

The program hangs in the call to set_rx_freq.

## Steps to reproduce the problem ##

1. Build v4.0.0.0-rc1
2. install
3. run rx_ascii_art_dft as above. 

## Additional Information ##

This worked in 3.15 and earlier.

michaelld, 2020-09-04:
@kb1vc thanks for reporting. We have tested / verified fix in the pipeline already, which will be part of the UHD 4.0 release. I'm going to close out this public issue; our internal issue will be resolved when the fix is merged.

closed 2020-09-04 by michaelld

kb1vc, 2020-09-04:
great.  Will it be in a release candidate first?  I'd like to try SoDaRadio against it, and this bug blocks testing.  It would be helpful if I could get a little ahead of the SoDaRadio users: there aren't many of them, but I try to answer issues from them when I can.

michael-west, 2020-09-06:
@kb1vc  The fix is now on the master and UHD-4.0 branches.

kb1vc, 2020-09-08:
and it works with SoDaRadio.  2655b0aa4abf9cba79b1f160e025909b7c5d598c

thanks.