EttusResearch/uhd #371: set_rx_rate()  takes 300ms-500ms.

commthu, 2020-08-31:
My application require switch sample rate very often,  However set_rx_rate() takes 300ms-500ms time.
How can i reduce this overhead.

michaelld, 2020-08-31:
which USRP are you using here?

commthu, 2020-08-31:
B210 mini

michaelld, 2020-09-01:
That's useful information. Some other useful information would include your host computer make/model, and the USB connection (for example "lsusb" if you're on Linux).

This is not an issue with UHD, per se, so I'm going to close this out.

I encourage you to sign up and use the USRP-users email discussion list < https://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com >. You can try sending an email to "support@ettus.com", but they are overwhelmed & so you're much more likely to get a reply from USRP-users.

closed 2020-09-01 by michaelld