EttusResearch/uhd #372: RfnocError: OpTimeout: Control operation timed out waiting for space in command buffer

boogachamp, 2020-09-01:
## Issue Description ##
With RC v4.0.0.0
Built and ran benchmark_rate test:
./benchmark_rate --args="addr=192.168.40.2" --rx_rate=20e6 --pps gpsdo --ref gpsdo --channels "0,1" --random
majority of the time I die with:
terminate called after throwing an instance of 'uhd::op_timeout'
  what():  RfnocError: OpTimeout: Control operation timed out waiting for space in command buffer
Aborted (core dumped)

removing the --random usually results in fairly clean runs.

## Setup Details ##
CentOS 7, built with devtoolset 9
v4.0.0.0 RC
# ../utils/uhd_usrp_probe --args="addr=192.168.40.2"
[INFO] [UHD] linux; GNU C++ version 9.3.1 20200408 (Red Hat 9.3.1-2); Boost_106500; UHD_4.0.0.0-0-unknown
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [GPS] Found an internal GPSDO: LC_XO, Firmware Rev 0.929a
[INFO] [X300] Radio 1x clock: 200 MHz
  _____________________________________________________
 /
|       Device: X-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: X310
|   |   revision: 10
|   |   revision_compat: 7
|   |   product: 30818
|   |   mac-addr0: 00:80:2f:27:52:82
|   |   mac-addr1: 00:80:2f:27:52:83
|   |   gateway: 192.168.10.1
|   |   ip-addr0: 192.168.10.2
|   |   subnet0: 255.255.255.0
|   |   ip-addr1: 192.168.20.2
|   |   subnet1: 255.255.255.0
|   |   ip-addr2: 192.168.30.2
|   |   subnet2: 255.255.255.0
|   |   ip-addr3: 192.168.40.2
|   |   subnet3: 255.255.255.0
|   |   serial: 30F9772
|   |   FW Version: 6.0
|   |   FPGA Version: 38.0
|   |   FPGA git hash: fffffff-dirty
|   |
|   |   Time sources:  internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: gps_servo, gps_gprmc, gps_time, gps_gpgga, gps_locked, ref_locked
|     _____________________________________________________
|    /
|   |       RFNoC blocks on this device:
|   |
|   |   * 0/DDC#0
|   |   * 0/DDC#1
|   |   * 0/DUC#0
|   |   * 0/DUC#1
|   |   * 0/Radio#0
|   |   * 0/Radio#1
|     _____________________________________________________
|    /
|   |       Static connections on this device:
|   |
|   |   * 0/SEP#0:0==>0/DUC#0:0
|   |   * 0/DUC#0:0==>0/Radio#0:0
|   |   * 0/Radio#0:0==>0/DDC#0:0
|   |   * 0/DDC#0:0==>0/SEP#0:0
|   |   * 0/Radio#0:1==>0/DDC#0:1
|   |   * 0/DDC#0:1==>0/SEP#1:0
|   |   * 0/SEP#2:0==>0/DUC#1:0
|   |   * 0/DUC#1:0==>0/Radio#1:0
|   |   * 0/Radio#1:0==>0/DDC#1:0
|   |   * 0/DDC#1:0==>0/SEP#2:0
|   |   * 0/Radio#1:1==>0/DDC#1:1
|   |   * 0/DDC#1:1==>0/SEP#3:0
|     _____________________________________________________
|    /
|   |       TX Dboard: dboard
|   |   ID: UBX-160 v2 (0x007d)
|   |   Serial: 310D1C2
|   |     _____________________________________________________
|   |    /
|   |   |       TX Frontend: 0
|   |   |   Name: UBX TX
|   |   |   Antennas: TX/RX, CAL
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 10.000 to 6000.000 MHz
|   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   Bandwidth range: 160000000.0 to 160000000.0 step 0.0 Hz
|   |   |   Connection Type: QI
|   |   |   Uses LO offset: No
|     _____________________________________________________
|    /
|   |       RX Dboard: dboard
|   |   ID: UBX-160 v2 (0x007e)
|   |   Serial: 310D1C2
|   |     _____________________________________________________
|   |    /
|   |   |       RX Frontend: 0
|   |   |   Name: UBX RX
|   |   |   Antennas: TX/RX, RX2, CAL
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 10.000 to 6000.000 MHz
|   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   Bandwidth range: 160000000.0 to 160000000.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No
|     _____________________________________________________
|    /
|   |       TX Dboard: dboard
|   |   ID: UBX-160 v2 (0x007d)
|   |   Serial: 310D168
|   |     _____________________________________________________
|   |    /
|   |   |       TX Frontend: 0
|   |   |   Name: UBX TX
|   |   |   Antennas: TX/RX, CAL
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 10.000 to 6000.000 MHz
|   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   Bandwidth range: 160000000.0 to 160000000.0 step 0.0 Hz
|   |   |   Connection Type: QI
|   |   |   Uses LO offset: No
|     _____________________________________________________
|    /
|   |       RX Dboard: dboard
|   |   ID: UBX-160 v2 (0x007e)
|   |   Serial: 310D168
|   |     _____________________________________________________
|   |    /
|   |   |       RX Frontend: 0
|   |   |   Name: UBX RX
|   |   |   Antennas: TX/RX, RX2, CAL
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 10.000 to 6000.000 MHz
|   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   Bandwidth range: 160000000.0 to 160000000.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No

## Expected Behavior ##
To not terminate

## Actual Behaviour ##
terminate called after throwing an instance of 'uhd::op_timeout'
  what():  RfnocError: OpTimeout: Control operation timed out waiting for space in command buffer
Aborted (core dumped)

## Steps to reproduce the problem ##
./benchmark_rate --args="addr=192.168.40.2" --rx_rate=20e6 --pps gpsdo --ref gpsdo --channels "0,1" --random

atrnati, 2020-09-24:
Would you be able to try reproducing the issue you report with the release tagged version of UHD 4.0.0 (or the head of `master`) and report your findings back to us?

michael-west, 2020-10-12:
The --random option is not properly implemented.  It is really not a valid use case, though.  It creates a random sized frame for every recv() call, which is not what gets produced when streaming data.  However, I was able to fix the issue with the following change:

```
diff --git a/host/examples/benchmark_rate.cpp b/host/examples/benchmark_rate.cpp
index 673649ddb..ddc0ea972 100644
--- a/host/examples/benchmark_rate.cpp
+++ b/host/examples/benchmark_rate.cpp
@@ -100,6 +100,11 @@ void benchmark_rx_rate(uhd::usrp::multi_usrp::sptr usrp,
     const double rate = usrp->get_rx_rate();
 
     uhd::stream_cmd_t cmd(uhd::stream_cmd_t::STREAM_MODE_START_CONTINUOUS);
+    cmd.num_samps = max_samps_per_packet;
+    if (random_nsamps) {
+        cmd.stream_mode = uhd::stream_cmd_t::STREAM_MODE_NUM_SAMPS_AND_DONE;
+        cmd.num_samps = (rand() % (max_samps_per_packet - 1)) + 1;
+    }
     cmd.time_spec  = usrp->get_time_now() + uhd::time_spec_t(rx_delay);
     cmd.stream_now = (buffs.size() == 1);
     rx_stream->issue_stream_cmd(cmd);
@@ -115,11 +120,12 @@ void benchmark_rx_rate(uhd::usrp::multi_usrp::sptr usrp,
             stop_called = true;
         }
         if (random_nsamps) {
-            cmd.num_samps = rand() % max_samps_per_packet;
+            cmd.time_spec  = usrp->get_time_now() + uhd::time_spec_t(rx_delay);
+            cmd.num_samps = (rand() % (max_samps_per_packet - 1)) + 1;
             rx_stream->issue_stream_cmd(cmd);
         }
         try {
-            num_rx_samps += rx_stream->recv(buffs, max_samps_per_packet, md, recv_timeout)
+            num_rx_samps += rx_stream->recv(buffs, cmd.num_samps, md, recv_timeout)
                             * rx_stream->get_num_channels();
             recv_timeout = burst_pkt_time;
         } catch (uhd::io_error& e) {
```

michael-west, 2020-10-12:
The fix is in review and will be available on the master branch soon.  Thank you!

closed 2020-10-12 by michael-west