EttusResearch/uhd #375: Boost not Found?

tmautry, 2020-09-04:
## Issue Description ##
Just got a USRP B210 and am trying to install on windows the UHD software to talk to it. Have never used ettus or UHD products before. Due to other hardware, am using microsoft visual studio 2019 and an older version of libusb.

Have followed the instructions found here:
https://kb.ettus.com/Building_and_Installing_the_USRP_Open_Source_Toolchain_(UHD_and_GNU_Radio)_on_Windows#Doxygen

Using the most up-to-date binaries.

 Check the Advanced checkbox
 Click Configure
 Set Visual Studio 16 2019 Win64 as the compiler
 Click “Finish” and allow CMake to Generate

## Setup Details ##
I have boost installed at C:\Program Files\Boost\boost_1_74_0

I copied all of the github distribution to the following directory (I don't use/know how to use git )
C:\Users\...\uhd-release\host\uhd-master

## Expected Behavior ##

## Actual Behaviour ##
Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.17763.

Configuring the Python interpreter...
Manually determining build Python version...
Python interpreter: C:/Users/tmautry/Anaconda3/python.exe Version: 3.8.3
Override with: -DPYTHON_EXECUTABLE=<path-to-python>
Manually determining runtime Python version...
Python runtime interpreter: C:/Users/tmautry/Anaconda3/python.exe Version: 3.8.3
Override with: -DRUNTIME_PYTHON_EXECUTABLE=<path-to-python>
Finding Python Libraries...
Python Libraries: C:/Users/tmautry/Anaconda3/libs/python38.lib
Python include directories: C:/Users/tmautry/Anaconda3/include
CMake Warning at cmake/Modules/UHDVersion.cmake:65 (message):
  Could not detect git executable! Could not determine exact version of UHD!
Call Stack (most recent call first):
  cmake/Modules/UHDPackage.cmake:9 (include)
  CMakeLists.txt:103 (include)

Using UHD Images Directory: C:\Program Files (x86)\UHD\share\uhd\images

Configuring Boost C++ Libraries...

Checking for Boost version 1.58 or greater
  Looking for required Boost components...
CMake Error at C:/Program Files/CMake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR) (Required is at least
  version "1.58")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.18/Modules/FindBoost.cmake:2177 (find_package_handle_standard_args)
  cmake/Modules/UHDBoost.cmake:176 (find_package)
  CMakeLists.txt:296 (include)

Configuring incomplete, errors occurred!
See also "C:/Users/tmautry/uhd-release/host/uhd-master/host/build/CMakeFiles/CMakeOutput.log".

## Steps to reproduce the problem ##

## Additional Information ##

mbr0wn, 2020-09-07:
@tmautry Is there a reason you can't use the pre-compiled binaries? Boost and MSVC is always a bit complicated. You have to set a lot of CMake variables manually. I assume you followed the build instructions on the knowledge base?

tmautry, 2020-09-08:
If you can point me to a working list of pre-compiled binaries that would be great.

Your instructions

https://kb.ettus.com/B200/B210/B200mini/B205mini_Getting_Started_Guides

Tell me to create the binaries by sending me to

https://kb.ettus.com/Building_and_Installing_the_USRP_Open_Source_Toolchain_(UHD_and_GNU_Radio)_on_Windows.

-----------------------
I have tried to install the binaries found here

https://files.ettus.com/binaries/uhd/latest_stable/Windows-10-x64/.

However, the next step is to do https://kb.ettus.com/Verifying_the_Operation_of_the_USRP_Using_UHD_and_GNU_Radio

This is a bunch of linux commands with no windows documentation.  So I have no idea how to do it since I am on windows.

Additionally, looking at the USB drivers found here http://files.ettus.com/binaries/misc/erllc_uhd_winusb_driver.zip

There is no uhd_b210 and so I don’t know what I am supposed to do for a driver etc.

-----------------------------

Finally, even when I have installed the application using the binaries found
https://files.ettus.com/binaries/uhd/latest_stable/Windows-10-x64/.

I do not know how to launch a UHD or GNU Radio application since there is no installed GNU Radio or UHD application?

Best,
Travis

CONFIDENTIALITY NOTICE: The information transmitted in this email, including attachments, is intended only for the person(s) or entity to which it is addressed and may contain confidential, proprietary and/or privileged material exempt from disclosure under applicable law. Any review, retransmission, dissemination or other use of, or taking of any action in reliance upon this information by persons or entities other than the intended recipient is prohibited. If you received this message in error, please contact the sender immediately and destroy any copies of this information in their entirety.

mbr0wn, 2020-09-09:
On 9/8/20 6:25 PM, tmautry wrote:
> -----------------------
> I have tried to install the binaries found here
> 
> https://files.ettus.com/binaries/uhd/latest_stable/Windows-10-x64/.

Those are fine.

> However, the next step is to do
> https://kb.ettus.com/Verifying_the_Operation_of_the_USRP_Using_UHD_and_GNU_Radio
> 
> This is a bunch of linux commands with no windows documentation. So I
> have no idea how to do it since I am on windows.

These commands will work just fine on Windows. The page you point to has
Windows-specific instructions.

> Additionally, looking at the USB drivers found here
> http://files.ettus.com/binaries/misc/erllc_uhd_winusb_driver.zip
> 
> There is no uhd_b210 and so I don’t know what I am supposed to do for a
> driver etc.

That zip files contains .inf files for all B2xx devices.

> -----------------------------
> 
> Finally, even when I have installed the application using the binaries
> found
> https://files.ettus.com/binaries/uhd/latest_stable/Windows-10-x64/.
> 
> I do not know how to launch a UHD or GNU Radio application since there
> is no installed GNU Radio or UHD application?

The UHD examples and utilities get installed as part of the binary
installer.

closed 2020-09-09 by mbr0wn

tmautry, 2020-09-09:
Please don't close the help request before it is finished.  Or if that is really the attitude then please consider sending me an RMA.  I have spent several hours trying to get your product to work and am asking for help because I am unfamiliar with it and your documentation is lacking.  More than a one sentence response would be appreciated.

-----

> However, the next step is to do [...]

These commands will work just fine on Windows. The page you point to has
Windows-specific instructions.

First: There are not enough instructions to get it to work on windows if unfamiliar with your product.  The only mention of windows specific instructions on that page are here

[cid:2580bd73-059e-4ca0-bfaa-4b37b846d481]

Those commands do not work in windows.  See for instance in a terminal when I run

uhd_usrp_probe.

(base) C:\Users\tmautry>uhd_usrp_probe
'uhd_usrp_probe' is not recognized as an internal or external command,
operable program or batch file.

-----------

Here are the contents of the USB folder that I indicated to you.  Can you please tell me which one of these is appropriate for the B210 which is not listed?

[cid:dae9233d-1670-4024-bfe8-1d86fe6aee8d]

michaelld, 2020-09-09:
@tmautry The GitHub issue tracker is for UHD and USRP-based issues -- which can include build problems That said, our internal buildbots are working just fine and hence your issue here is with the Ettus KB / other documentation / that you're just having issues building UHD on your particular OS -- which are all outside the scope of this issue tracker. This is why Martin closed the issue, and why we won't reopen it -- unless you can show that the UHD build system has a bug in it; installing dependencies to build UHD is outside the scope, too. My recommendation is to send an email to "support@ettus.com" and we'll try to help. I can't guarantee a quick reply given our work queues and a conference next week, but we will get to your query. Ettus Support is the right place to ask for help with UHD installation, if there is a place in Ettus to be asking.

mbr0wn, 2020-09-09:
> In a terminal when I run uhd_usrp_probe I get the following error:
> 
> (base) C:\Users\tmautry>uhd_usrp_probe
> 'uhd_usrp_probe' is not recognized as an internal or external command,
> operable program or batch file.
> 
> Finally there are no windows specific instructions on that wiki page other than.
> 
> **On Windows, the default installation prefix will be C:\Program Files\UHD\.**
> **The example programs will be located in the folder:**
> **C:\Program Files\UHD\lib\uhd\examples**

@tmautry please refer to what @michaelld said. Apologies if this came over curt, but we'll try and close out support issues (that are not bugs) on the bug tracker quickly so they don't get conflated with actual bugs.
In your last response, you mention that you tried to execute `uhd_usrp_probe` from `C:\Users\tmautry`. However, further down, you mention that the examples are actually in the directory `C:\Program Files\UHD\lib\uhd\examples`. You can also use the Windows search function to verify the actual path; it might differ on your system. Note that the the KB page is mostly OS-agnostic, all the commands work identically on Windows, Mac, and Linux. Most of our customers prefer Linux, so we gear the tutorials toward that majority. The specific page you mention has nothing Linux-specific, except for a mention of `sudo`, which is not required on Windows, and some paths. I have modified the page accordingly.

It helps us if you provide some level of proficiency. I realize my response was unhelpful/perceivable as glib if you're not actually familiar with PATHs etc. So, poke support, and we'll get you straightened out. We just need to remove this from the bug tracker.