EttusResearch/uhd #376: Missing cmrc on build

SpontaneousDuck, 2020-09-14:
## Issue Description ##
Building new 4.0 from source gives `fatal error: cmrc/cmrc.hpp: No such file or directory` when running `make`.

## Setup Details ##

Tag: v4.0.0.0
Hash: 90ce6062b6b5df2eddeee723777be85108e4e7c7
OS: Ubuntu 18.04

## Expected Behavior ##

Successful build or fail if I need a dependency.

## Actual Behaviour ##

```
uhd/lib/cal/database.cpp:12:10: fatal error: cmrc/cmrc.hpp: No such file or directory
 #include <cmrc/cmrc.hpp>
          ^~~~~~~~~~~~~~~
compilation terminated.
lib/CMakeFiles/uhd_static.dir/build.make:215: recipe for target 'lib/CMakeFiles/uhd_static.dir/cal/database.cpp.o' failed
make[2]: *** [lib/CMakeFiles/uhd_static.dir/cal/database.cpp.o] Error 1
CMakeFiles/Makefile2:121: recipe for target 'lib/CMakeFiles/uhd_static.dir/all' failed
make[1]: *** [lib/CMakeFiles/uhd_static.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
```
## Steps to reproduce the problem ##

1. Download source tarball from release on Github [https://github.com/EttusResearch/uhd/archive/v4.0.0.0.tar.gz](https://github.com/EttusResearch/uhd/archive/v4.0.0.0.tar.gz)
2. Build boost with dependencies and set `$EXT_BUILD_DEPS` to where these exist
3. ```
mkdir host/build
cd host/build
cmake -DBOOST_ROOT="$EXT_BUILD_DEPS/boost/" ../
make
```

## Additional Information ##
<!--- [Any additional information, configuration or data that might be necessary to reproduce the issue] -->

michaelld, 2020-09-14:
What version of CMake are you using?

SpontaneousDuck, 2020-09-14:
Newest on Ubuntu package repository for bionic so `3.10.2`

SpontaneousDuck, 2020-09-14:
It appears like I get this error only when trying to build a static library now with ENABLE_STATIC_LIBS. I am not sure what changed but I was successfully able to build now if I do not use that flag.

mbr0wn, 2020-09-15:
@SpontaneousDuck Thanks for the report. You're right, it looks like the CMRC files aren't being built correctly for static builds.

SpontaneousDuck, 2020-09-15:
Perfect! I will continue with shared libs in the meantime

TSlivede, 2021-06-21:
The error seems to be fixed using [this](https://github.com/EttusResearch/uhd/commit/6a0c71826dd5ecfd1a4e870a558df3919f9fb827) workaround. (I compiled on Ubuntu 18.04)

@atrnati Will that commit be merged into the release branch?

pfeatherstone, 2021-08-05:
Is there an update on this?

pfeatherstone, 2021-08-05:
Also, when running ```sudo make install``` or ```sudo ninja install``, everything builds find, but weirdly, libuhd doens't get installed in ```/usr/local/lib```...

youkozlov, 2021-09-10:
Hi, I faced some problem with static lib building and it helped me:

host/lib/CMakeLists.txt
 if(ENABLE_STATIC_LIBS)
**+    list(APPEND libuhd_sources $<TARGET_OBJECTS:uhd_rc>)**
     add_library(uhd_static STATIC ${libuhd_sources})
**+    target_link_libraries(uhd_static cmrc::base)**
     set_target_properties(uhd_static PROPERTIES OUTPUT_NAME uhd)

mbr0wn, 2021-09-10:
Thanks -- we chose a similar approach to @youkozlov. We have a fix percolating our CI systems.

mbr0wn, 2021-09-10:
Most likely. Keep an eye on the master branch to see what's going to happen.

pfeatherstone, 2021-09-10:
cheers

closed 2021-09-16 by atrnati