EttusResearch/uhd #378: Fail to import the uhd Python module

starsway, 2020-09-16:
Hi,

## Issue Description ##
I would like to write programs using the UHD Python API. To this intent, I have built UHD from sources and everything went well, in the sense that 'LibUHD - Python API' is listed among 'UHD enabled components' by cmake, and compilation completes with no error. 'make install' is displaying the following lines:
`-- Installing: /usr/local/lib/python3/dist-packages/uhd/usrp.py
-- Installing: /usr/local/lib/python3/dist-packages/uhd/types.py
-- Installing: /usr/local/lib/python3/dist-packages/uhd/filters.py
-- Installing: /usr/local/lib/python3/dist-packages/uhd/__init__.py
-- Installing: /usr/local/lib/python3/dist-packages/uhd/libpyuhd.cpython-36m-x86_64-linux-gnu.so`

I find the name of the library libpyuhd.cpython-36m-x86_64-linux-gnu.so a bit surprising, but I'm not sure there is anything wrong about it.

Unfortunately, I cannot load the module 'uhd':

`$ python3
Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import uhd
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'uhd'
`
I made sure to execute ldconfig before trying. Python2 fails all the same.

## Setup Details ##

This happens under Ubuntu 18.04:
`$ cat /proc/version
Linux version 4.15.0-1096-oem (buildd@lcy01-amd64-023) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #106-Ubuntu SMP Sat Sep 5 15:59:05 UTC 2020`

I am using the latest sources from branch UHD-3.15.LTS:

`$ git log --pretty=oneline -5
bc585b4368ebdae3887bc1bcf4ba8fd7f52b25f0 (HEAD -> UHD-3.15.LTS, origin/UHD-3.15.LTS) utils: Make uhd_config_info print help by default
63210c2fc72b95ec6c40867c81e5a240f213828c utils: Fix prefix determination in get_lib_path()
1f47dc1b7b91f5c36cc349106ffe778262d1ae80 lib: Make sure generated file is closed
13caaf001061db3c01082c4574a5e326c4969ab6 boost: Include bind.hpp where used, add BOOST_BIND_GLOBAL_PLACEHOLDERS
bd00790199cd86bc16a47622ad37052995c69ae9 ad9361: Fix mask for product ID check`

I am not sure whether this is related to the name of the .so ; should it be libpyuhd.so ? Did cmake fail to name it properly?

`$ find /usr/ -name "libpyuhd*"
/usr/local/lib/python3/dist-packages/uhd/libpyuhd.cpython-36m-x86_64-linux-gnu.so
$ ls -l /usr/local/lib/python3/dist-packages/uhd/
total 1128
-rw-r--r-- 1 root root     402 sept. 16 14:13 filters.py
-rw-r--r-- 1 root root     211 sept. 16 14:13 __init__.py
-rw-r--r-- 1 root root 1130688 sept. 16 14:13 libpyuhd.cpython-36m-x86_64-linux-gnu.so
-rw-r--r-- 1 root root     983 sept. 16 14:13 types.py
-rw-r--r-- 1 root root    5112 sept. 16 14:13 usrp.py`

Any help or hint would be greatly appreciated. Thanks!

michaelld, 2020-09-16:
What does the following return: `python3 -c "import sys; print (sys.path)"` ... the UHD Python files must be in the `PYTHONPATH`, whether as set in the shell environment executing `python3` or within Python itself.

starsway, 2020-09-16:
Hi, thank you for your reply. Here is what the command you have suggested returns:

> $ python3 -c "import sys; print (sys.path)"
> ['', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/home/my_login/.local/lib/python3.6/site-packages', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages']

So I can see that '/usr/local/lib/python3.6/dist-packages' and '/usr/lib/python3/dist-packages' are listed, but not  '/usr/local/lib/python3/dist-packages/'... Is something broken in my python3 install...?

Thanks again.

michaelld, 2020-09-16:
You have a normal install, which is good! You just need to add something like the following line to the file `~/.bashrc` or `~/.bash_login` or whatever shell config file you use:
```
export PYTHONPATH=/usr/local/lib/python3/dist-packages:${PYTHONPATH}
```
Then, when you start a new shell if you do the following:
```
echo $PYTHONPATH
```
it should return a line that includes `/usr/local/lib/python3/dist-packages` ... and, if so, then when you execute `python3` you should now be able to `import uhd`.

starsway, 2020-09-16:
I confirm everything is working as you said - many thanks! I will close this issue, then.

closed 2020-09-16 by starsway

michaelld, 2020-09-16:
YW!